gyroflow/gyroflow #754: ffmpeg error: ffmpeg::Error(2: No such file or directory)

someTV, 2023-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

mac ventura 13.5.2

### What GPU are you using?

apple m1 max

### What happened?

in queue and expected a render to take place
[image]

AdrianEddy, 2023-11-09:
there are some issues with app sandbox, you can install from .dmg as a workaround for now https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-macos.zip

someTV, 2023-11-09:
Thanks, will try
Barnaby
323.401.8642

AdrianEddy, 2023-11-19:
Please try the TestFlight version and let me know if that fixes your issues: https://testflight.apple.com/join/e0RI1n5i

AdrianEddy, 2023-11-30:
it's fixed in the testflight version

closed 2023-11-30 by AdrianEddy

someTV, 2023-12-01:
using v1.5.4 (gh1679) and it is working.
Thanks
Barnaby
323.401.8642