gyroflow/gyroflow #755: not seeing "preserve other tracks" button in advanced encoding section

someTV, 2023-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4 (gh1663)

### What operating system are you using?

mac ventura 13.5.2

### What GPU are you using?

apple m1 max

### What happened?

Not seeing the preserve other tracks button in export settings advanced
[image]

AdrianEddy, 2023-11-09:
Click the blue "Advanced" link in Export settings panel

closed 2023-11-09 by AdrianEddy

someTV, 2023-11-10:
I see my error, when looking at the export settings and the "advanced" settings below made me think that the export "advance" settings were already open. Thanks for your help.