gyroflow/gyroflow #759: please keep screen on while export

DGYT, 2023-11-14:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Android export must keep screen on,but now screen will off while export, cause export pause

can keep screen on by 
getWindow().addFlags(WindowManager.LayoutParams.FLAG_KEEP_SCREEN_ON);

AdrianEddy, 2023-11-14:
Does it really interrupt the render for you? It shouldn't because Gyroflow acquires partial wake lock which should keep the CPU running, even if the screen goes off by itself: https://developer.android.com/training/scheduling/wakelock#cpu

DGYT, 2023-11-14:
as the video shows,it goes a little in near 1min（seams stopped）

https://github.com/gyroflow/gyroflow/assets/22211278/d630a6de-5b39-4548-8fde-5f894e537205

DGYT, 2023-11-14:
mistake close

AdrianEddy, 2023-11-16:
Please try latest apk https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-android.zip

AdrianEddy, 2023-11-16:
it works for me with the latest apk

closed 2023-11-16 by AdrianEddy