gyroflow/gyroflow #760: Unable to find Gyroflow app path. Make sure to run Gyroflow app at least once and that version is at least v1.4.3

pljeferson, 2023-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v.1.5.4

### What operating system are you using?

Mac OS - Sonoma 14.0

### What GPU are you using?

Apple Silicon M1

### What happened?

NLE: Davinci Resolve Studio 18.6.1 Build 8

After importing my clip into the Timeline and applying the OFX Plugin to the clip. After click for "Load for current file" and then when i try to "Open in Gyroflow" I can't access the program, the Gyroflow won't start even though I have already used it standalone before. 

[image]
I just get a message: "Unable to find Gyroflow app path. Make sure to run Gyroflow app at least once and that version is at least v1.4.3"

[image]
[image]

latenitefilms, 2023-11-21:
Do you have the main Gyroflow application in your `/Applications` folder?

AdrianEddy, 2023-11-21:
use latest dev build of the plugin from https://gyroflow.xyz/devbuild/
also if using plugin, install Gyroflow from .dmg instead of app store, because of app sandbox limitations