gyroflow/gyroflow #761: Gyroflow 1.5.4 cannot open

RyanFPVwing, 2023-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

windows 10

### What GPU are you using?

NVIDEA GEFORCE 940MX

### What happened?

Hi, I cannot launch a newly downloaded gyro.exe version 1.5.4.  I am using windows 10, updated geforce 940MX. I have already downloaded the missing .dll files.

AdrianEddy, 2023-11-23:
What's the problem?

RyanFPVwing, 2023-11-24:
Hi Adrian,

I need help setting up my gyroflow, I can't open the program with the
latest version.

AdrianEddy, 2023-11-24:
But what's the problem exactly? Is there any error message?

RyanFPVwing, 2023-11-24:
There is no error message, I just can't
 VID20231124151223[1].mp4
<https://drive.google.com/file/d/1EN024lyJfdso-T6m56TuRBgwMhB7ppQT/view?usp=drive_web>
 simply run the program. I think I have a compatibility issue, previously
my laptop was in windows 7, I encountered several missing dll. files, I
downloaded the missing files and updated my laptop to windows 10 and still
no progress. Gyroflow is my only way of stabilizing videos from my newly
bought hawkeye 4k thumb camera. Thank you.

AdrianEddy, 2023-11-24:
Install this https://aka.ms/vs/17/release/vc_redist.x64.exe

And install latest GPU drivers for your GPU, it's very important

RyanFPVwing, 2023-11-24:
I have already installed that software yesterday but it still does not
work. My GPU(geforce 940mx) is also updated.

AdrianEddy, 2023-11-24:
What's this laptop model?

RyanFPVwing, 2023-11-24:
Acer Aspire E 15 core i5

gecpogi, 2023-12-07:
I am having the same issue. Keeps on loading. I've already installed vc_redist64 and updated my graphics driver to the latest version. I'm using Acer Aspire 7 | Ryzen 5 | RTX3050 | 16gb RAM

Tried reinstalling many times with no luck. Tried the store version and the zip version. It's working fine on my pc but I can't get it to work with my laptop.

[image]

AdrianEddy, 2023-12-07:
@gecpogi this is most likely caused by the missing GPU drivers for your **other** GPU, which is integrated in your CPU. You need to install AMD drivers, it'll probably be this one https://drivers.amd.com/drivers/whql-amd-software-adrenalin-edition-23.11.1-win10-win11-nov3-vega-polaris.exe

gecpogi, 2023-12-08:
> @gecpogi this is most likely caused by the missing GPU drivers for your **other** GPU, which is integrated in your CPU. You need to install AMD drivers, it'll probably be this one https://drivers.amd.com/drivers/whql-amd-software-adrenalin-edition-23.11.1-win10-win11-nov3-vega-polaris.exe

Thank you so much. It's working now. :)

AdrianEddy, 2023-12-13:
@ferdiaz63 Install this driver https://us.download.nvidia.com/Windows/391.35/391.35-notebook-win10-64bit-international-whql.exe

closed 2023-12-13 by AdrianEddy

RyanFPVwing, 2023-12-16:
Thank you @AdrianEddy , I can now open Gyroflow and stabilize my videos, however, I came up with a new problem, I cannot save or export the rendered video clips. 
[image]

RyanFPVwing, 2023-12-16:
Thank you @AdrianEddy <https://github.com/AdrianEddy> , I can now open
Gyroflow and stabilize my videos, however, I came up with a new problem, I
cannot save or export the rendered video clips.

AdrianEddy, 2023-12-16:
@RyanFPVwing change export codec to H264 or disable GPU encoding

RyanFPVwing, 2023-12-16:
works like a miracle, thank you @AdrianEddy and to the gyroflow team.