gyroflow/gyroflow #762: opencamera sensors motion data is not detected on latest android dev build

nex3n, 2023-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

android latest

### What operating system are you using?

android 11

### What happened?

opencamera sensors motion data is not detected on latest android dev build

nex3n, 2023-11-26:
[sample.zip](https://github.com/gyroflow/gyroflow/files/13467722/sample.zip)

AdrianEddy, 2023-11-30:
This is unlikely to be supported anytime soon. Due to Android's complicated filesystem access and permissions, it would require a lot of additional work to support that in the gyro parsing library, which is not something I plan doing for this niche use case.

AdrianEddy, 2023-12-22:
It's done in latest dev build https://play.google.com/apps/testing/xyz.gyroflow

closed 2023-12-22 by AdrianEddy