gyroflow/gyroflow #763: MacOS custom lens profile from BRAW crash

Mcrubens, 2023-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

macOS Sonoma 14.1.1

### What GPU are you using?

M2 MacBook Pro 13" 2022

### What happened?

I'm trying to create a new lens profile from a BRAW video. Everything seems to work (see uploaded screenshot) until auto calibration reaches 100%. Then Gyroflow crashes.

[image]

Mcrubens, 2023-11-26:
This is the crash log from MacOS

[Crash log Gyroflow Custom lens profile.txt](https://github.com/gyroflow/gyroflow/files/13467948/Crash.log.Gyroflow.Custom.lens.profile.txt)

AdrianEddy, 2023-11-27:
It's fixed in latest dev build https://gyroflow.xyz/devbuild

closed 2023-11-27 by AdrianEddy