gyroflow/gyroflow #769: Set settings profile as default or add window with presets

abramowv, 2023-12-13:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I often stabilise gopro 8:7 videos twise, horizontal format and vertical. Please make settings preset automaticly applied when you open program or add window with presets with one click activation

AdrianEddy, 2023-12-13:
It's already there, just add your preset to camera_presets folder and it will appear in the lens profile list. You can then make it favorite and it will stay on top
https://docs.gyroflow.xyz/app/advanced-usage/settings-presets

closed 2023-12-13 by AdrianEddy

abramowv, 2023-12-13:
[image]

AdrianEddy, 2023-12-13:
Right, fixed in latest dev build https://gyroflow.xyz/devbuild/