gyroflow/gyroflow #771: Export problem, I used prores and h264/5,

Stef2365, 2023-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

Windows 10

### What GPU are you using?

Rtx3060 laptop

### What happened?

color layers shorter than brightness
[image]

AdrianEddy, 2023-12-20:
can you send me the original video file?

AdrianEddy, 2023-12-21:
Please try latest dev build https://gyroflow.xyz/devbuild/

closed 2024-01-18 by AdrianEddy