gyroflow/gyroflow #775: Crash on 'automatic lens calibration'

ChristosV22, 2023-12-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

macOS Sonoma 14.2.1

### What GPU are you using?

AMD Radeon Pro 5500M 8 GB

### What happened?

Gyroflow crashes once hitting the 'auto lens calibration' button in the lens calibrator window when trying to create a lens profile with a 6k Open Gate .braw file. I noticed [hawk20002000 posted](https://github.com/gyroflow/gyroflow/issues/660) a similar report but wasn't sure if it ever was resolved? I've tried the suggestions that were given in that post, with no luck.

AdrianEddy, 2023-12-27:
That's already fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2023-12-27 by AdrianEddy

ChristosV22, 2023-12-27:
Yes I downloaded that version and am encountering the same issue. Crash report lists it as a graphics hardware error.

AdrianEddy, 2023-12-28:
Please include gyroflow.log file (`Open package contents -> Contents -> MacOS`)

ChristosV22, 2023-12-28:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/13782171/gyroflow.log)

AdrianEddy, 2023-12-28:
can you also send me that .braw file?

ChristosV22, 2023-12-28:
Uploading now will include google drive link shortly

ChristosV22, 2023-12-28:
https://drive.google.com/file/d/1iZKRaylUh9ZeCSLWDf0glvdIKphyYZRk/view?usp=sharing

AdrianEddy, 2023-12-28:
Hmm it's working fine for me on Windows, I'll check on macOS tomorrow
In the meantime, here's the profile: 
[Blackmagic_Cinema Camera 6K_50mm__6k_3by2_6048x4032-24.00fps.json](https://github.com/gyroflow/gyroflow/files/13782987/Blackmagic_Cinema.Camera.6K_50mm__6k_3by2_6048x4032-24.00fps.json)

ChristosV22, 2023-12-28:
Thanks so much

AdrianEddy, 2023-12-28:
It may be GPU memory problem, can you check if it works if you lower the `Processing resolution`? It's in Advanced in the calibrator

ChristosV22, 2023-12-28:
I tried it at 720p resolution with the same result, however I am running on an older generation (2019) Macbook Pro so it may very well be a GPU issue.