gyroflow/gyroflow #776: SizeMismatch error in latest Gyroflow (1.5.4)

padcom, 2023-12-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

Linux padcom 6.2.0-37-generic #38~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov  2 18:01:13 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

### What GPU are you using?

RTX 3060 Ti

### What happened?

No stabilization, the preview is extremely chunky, nothing works as expected.

### Relevant log output

```shell
18:51:04 [INFO] Initialized OpenCL for (1920, 1080, 7680) -> (1920, 1080, 7680)
18:51:04 [INFO] Clearing 1 stab data because backend was initialized
18:51:05 [DEBUG] (26) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") Ok("OpenCL 3.0 CUDA 12.3.99") ["NVIDIA GeForce RTX 3060 Ti"]
18:51:05 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd" } Device: NVIDIA Corporation NVIDIA GeForce RTX 3060 Ti
18:51:05 [INFO] Initialized OpenCL for (1920, 1080, 7680) -> (1920, 1080, 7680)
18:51:05 [INFO] Clearing 1 stab data because backend was initialized
18:51:05 [ERROR] Failed to process pixels: SizeMismatch((1920, 1080), (0, 0))
```

padcom, 2023-12-28:
The issue does not appear in 1.5.2

AdrianEddy, 2023-12-28:
Did you try latest dev build? https://gyroflow.xyz/devbuild/

padcom, 2023-12-28:
Just the latest release - let me see the latest dev build

padcom, 2023-12-28:
Just tested the latest dev build and the bug does not appear to be there anymore. Thanks! Closing.

closed 2023-12-28 by padcom