gyroflow/gyroflow #778: Gyroflow 1.5.4 & Resolve 16.5.4 - GoPro orientation issue

alexrjohnston, 2024-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

osx - multiple

### What GPU are you using?

mac

### What happened?

If a GoPro is used upside down, with orientation locked so the footage appears to be the correct way up in GF, GF correctly detects that the GoPro is 180 degrees upside down - sadly if the GF file is imported into the OpenFX plugin in resolve then because davinci has also corrected which way is up the gyro data is out by 180 degrees

AdrianEddy, 2024-01-05:
This is what "Input rotation" is for in the plugin.
You can either:
a) go to Clip Attributes and set the rotation to 0
b) adjust "Input rotation" in plugin (and possibly Video rotation as well)

alexrjohnston, 2024-01-05:
Ah thanks, I thought it was intended to be able to correct automatically, if this is expected behaviour then happy days 👍🏻 thanks for the response

AdrianEddy, 2024-01-05:
Unfortunately there's no way for plugin to know what did Resolve do to file before feeding to the plugin so that's why you have to enter that manually using the "Input rotation".

It's important to know that Gyroflow plugin gets **timeline** pixels and not original file pixels, and since the way it works requires original untouched pixels from file, any manipulation on the pixels before Gyroflow breaks everything.

Resolve doesn't provide any API to know any transformations done to the pixels, so you have to be careful and provide all that manually

closed 2024-01-05 by AdrianEddy

alexrjohnston, 2024-01-06:
Thanks for the response and all your hard work on this project mate 👍🏻 It's brilliant software. 

Would there be a downside to gyroflow always assuming resolve has rotated any GoPro footage to be the right way up as I think that it always does? This would prevent the footage being flipped upside down again when the gyroflow project is loaded into the openfx plugin.