gyroflow/gyroflow #780: Better scale quality?

temp-64GTX, 2024-01-07:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Good evening. Me again.

Is it possible it implement a better scale quality option?
Typical case: i have a 4k video, open'd it in Gyroflow, apply stabilisation, and export to a 720p ProRes proxy, for fast editing. And this 720p video looks very "aliased". Like in games, you know, when you turn AA completely off. 
For example, this picture. 4k video. 
Upper one exported from Gyroflow in 720p. 
Lower one exported from Gyroflow in native 4k, and downscaled it video-editor. It's standard scale-processing, "bilinear", or something.
So we can clearly see the difference. Post (tall metal white tiny thing) in the center - upper one has this typical "stairs-edges". Same "stairs-edges" we can see on the wires, and on the vertical lines of beige building at the right. Meanwhile on the lower picture - all these lines are softer and smoother.

[image]

And in dynamic video it is much more visible, than in a static picture. 
So i think better scale quality will be great. At least for exporting videos. 
I guess bilinear works fine. Also i am using "Lanczos" in XnView for scaling pictures - it gives a good quality.

AdrianEddy, 2024-01-07:
The export scaling in Gyroflow is Lanczos4 already

temp-64GTX, 2024-01-07:
Here is video-comparison. Youtube eats video-quality, but difference are still visible. 

https://www.youtube.com/watch?v=myNK5hb4Wek

AdrianEddy, 2024-01-07:
Added the selector in [7b48ed3](https://github.com/gyroflow/gyroflow/commit/7b48ed3c120bea780f8c2f64720093c2d015f1a0), in Export settings -> Advanced

temp-64GTX, 2024-01-08:
Selector is great, but nothing is changed :(
All 3 methods giving almost the same results with "stairs-edges" :(

AdrianEddy, 2024-01-08:
yeah that's why I think there's something else going on, please do more testing:
- different export codecs
- using OpenFX plugin in Fusion, with export resolution (in gyroflow project) set to 720p
- render at original resolution but set FOV to 4 to simulate upscaling

temp-64GTX, 2024-01-09:
Short Results:

All of different (png, prores, h265, etc.) export codecs with 720p : stairs-edges.
4k & FOV 4: stairs-edges.
4k & FOV 0.25: Smooth Edges.
DaVinci Resolve: crashed three times on exporting process. Then i just removed it to hell, where it came from. "use DaVinci", they said. "it fast and simple", they said.

[image]

temp-64GTX, 2024-01-11:
Ok, i tried Fusion Studio.
it's more or less working, but i can't understand how to tell the gyroflow that it must do a downscale. Because now it is Fusion doing it.

[image]

temp-64GTX, 2024-01-11:
Yep. It's also giving the stairs-edges.

AdrianEddy, 2024-01-11:
Can you send me the sample file?

temp-64GTX, 2024-01-11:
i hope link is working. In the video, this effect will be very noticeable at the beginning - bridge lines and fence lines.
https://drive.google.com/file/d/1QrcdP1I8hhMzMIARPLLUempavVZq__LO/view?usp=sharing

temp-64GTX, 2024-01-23:
Meanwhile, i've tested mobile version of gyroflow. To exclude any possibility of impact the result by my particular pc-hardware. 
(And result is same)
I put original gopro file and all 3 output files to the link above.

hawku, 2024-04-26:
I have the same problem with GoPro videos. It looks like there isn't any filters used for scaling.
Scaling from 4k to 8k looks good but from 4k to 720p is completely unusable.
I have to export 8k and scale that down with ffmpeg to get decent output from Gyroflow.

8k export and scaled to 720p:
[image]

720p export:
[image]

AdrianEddy, 2024-04-26:
I still can't reproduce this issue, the 720p exports look perfectly fine for me, and of course there are filters used when exporting, and high quality Lanczos4 is the default

hawku, 2024-04-26:
Are you using high enough FOV?
Examples from footage provided by @temp-64GTX with FOV set to 3

8k export to 720p:
[image]

720p export:
[image]

AdrianEddy, 2024-04-26:
Ok I see it
Since our scaling code is ported from OpenCV, we've inherited the scaling algorithm that is not entirely correct:
- https://zuru.tech/blog/the-dangers-behind-image-resizing
- https://github.com/assafshocher/ResizeRight/
- https://richzhang.github.io/antialiased-cnns/

We should look into changing our scaling code to be based on pillow's: https://github.dev/zurutech/pillow-resize/blob/4427c50425bcdacffb129cce28d44afcf3ce0856/src/PillowResize/PillowResize.cc
They are pretty similar from the first glance, so looks like we need to dig deeper and identify what's exactly different. 
Not trivial, but also not terrible

AdrianEddy, 2024-04-26:
/bounty $500

AdrianEddy, 2024-05-01:
I've made some progress, I re-implemented the pillow's algorithm in Rust with a minimal example, comparing OpenCV's implementations (the code currently in Gyroflow) and Pillow's implementation.

The Pillow's one resizes the images better, however, the way it's structured is that it calculates coefficients up front for resizing from input to output. This is a problem, because in Gyroflow we need to feed input coordinates to the resampling function, because they will be rotated (stabilized), so it's different than simple resizing (where coordinates map in a simple linear way from input to output)

Example project: 
[resizing.zip](https://github.com/gyroflow/gyroflow/files/15182052/resizing.zip)

there's `pillow:sample_at_output`, which uses precomputed coefficients, but we can't use it as our source image coordinates can't be precomputed (because they are calculated in the gpu kernel per pixel)

I've implemented `pillow::sample_input_at`, but this one doesn't use any precomputed coeffcients, and instead calls the resampling function right there for every sampled pixel, which will be slow. This function also has a bug that it shifts the image a bit (compare 128.png output size 32 or 1024, with outputs from `test.py`)

I'm reducing the bounty to $150 since most of the work is done, just need to sort this one bug and figure out a way how to precompute some coefficients for that use case

/bounty $150

AdrianEddy, 2024-05-01:
/bounty $150

AdrianEddy, 2024-05-02:
I implemented it without precomputed coefficients in order to test, and the rendered video is now much nicer, check out the files there: https://drive.google.com/drive/folders/1brliOo0b4RLHOKbhraUBvIyRtsMIS-uj?usp=sharing

However, this improves things only when downscaling (massive improvement), but for regular videos (where there is mostly slight upscaling - see `__GH011230_stabilized.mp4`), I don't see any difference, and I've been pixel peeping pretty hard

This implementation is 2-3x slower to render

temp-64GTX, 2024-05-02:
Well, stairs is gone. However, new one is kind of blurry. 
[image]
(left - AfterEffects downscale, right - new downloaded from your google-drive)

AdrianEddy, 2024-05-02:
indeed
hmm, maybe this really needs to be a 2-stage process, first stabilize and then resize
doing it in one step sounds good on paper, but apparently has it's limitations

AdrianEddy, 2024-05-04:
For reference, here's the wgsl implementation if anyone wants to play with it
```wgsl
fn bilinear_filter(x_: f32) -> f32 { let x = abs(x_); if x < 1.0 { return 1.0 - x; } else { return 0.0; } }
fn hamming_filter(x_: f32) -> f32 { var x = abs(x_); if x == 0.0 { return 1.0; } else if x >= 1.0 { return 0.0; } else { x = x * 3.14159265359; return (sin(x) / x) * (0.54 + 0.46 * cos(x)); } }
fn bicubic_filter(x_: f32) -> f32 { let x = abs(x_); let A: f32 = -0.5; if x < 1.0 { return ((A + 2.0) * x - (A + 3.0)) * x * x + 1.0; } else if x < 2.0 { return (((x - 5.0) * x + 8.0) * x - 4.0) * A; } else { return 0.0; } }
fn sinc_filter(x: f32) -> f32 { if x == 0.0 { return 1.0; } else { let xx = x * 3.14159265359; return sin(xx) / xx; } }
fn lanczos_filter(x: f32) -> f32 { if x >= -3.0 && x < 3.0 { return sinc_filter(x) * sinc_filter(x / 3.0); } else { return 0.0; } }

fn sample_input_at2(uv_param: vec2<f32>) -> vec4<f32> {
    let filter_support = 3.0;
    let scale = min(params.fov, 10.0);
    let filter_scale = max(scale, 1.0);
    let support = filter_support * filter_scale;
    let ss = 1.0 / filter_scale;
    var kx = array<f32, 64>();
    var ky = array<f32, 64>();

    let fix_range = bool(flags & 1);

    let bg = params.background * params.max_pixel_value;
    var sum = vec4<f32>(0.0);

    var uv = uv_param;
    if (params.input_rotation != 0.0) {
        uv = rotate_point(uv, params.input_rotation * (3.14159265359 / 180.0), vec2<f32>(f32(params.width) / 2.0, f32(params.height) / 2.0));
    }

    if (bool(flags & 32)) { // Uses source rect
        uv = vec2<f32>(
            map_coord(uv.x, 0.0, f32(params.width),  f32(params.source_rect.x), f32(params.source_rect.x + params.source_rect.z)),
            map_coord(uv.y, 0.0, f32(params.height), f32(params.source_rect.y), f32(params.source_rect.y + params.source_rect.w))
        );
    }

    ////////////////////////////////
        let xcenter = uv.x + 0.5 * scale;
        let xmin = i32(floor(max(xcenter - support, 0.0)));
        let xmax = max(i32(ceil(min(xcenter + support, f32(params.width)))) - xmin, 0);
        var xw = 0.0;
        for (var x: i32 = 0; x < xmax; x = x + 1) {
            let f: f32 = (f32(x) + f32(xmin) - xcenter + 0.5) * ss;
            kx[x] = lanczos_filter(f);
            xw += kx[x];
        }
        if (xw != 0.0) { for (var x: i32 = 0; x < xmax; x = x + 1) { kx[x] /= xw; } }
    ////////////////////////////////
        let ycenter = uv.y + 0.5 * scale;
        let ymin = i32(floor(max(ycenter - support, 0.0)));
        let ymax = max(i32(ceil(min(ycenter + support, f32(params.height)))) - ymin, 0);
        var yw = 0.0;
        for (var y: i32 = 0; y < ymax; y = y + 1) {
            let f: f32 = (f32(y) + f32(ymin) - ycenter + 0.5) * ss;
            ky[y] = lanczos_filter(f);
            yw += ky[y];
        }
        if (yw != 0.0) { for (var y: i32 = 0; y < ymax; y = y + 1) { ky[y] /= yw; } }
    ////////////////////////////////

    let sx = xmin;
    let sy = ymin;

    for (var yp: i32 = 0; yp < ymax; yp = yp + 1) {
        if (sy + yp >= params.source_rect.y && sy + yp < params.source_rect.y + params.source_rect.w) {
            var xsum = vec4<f32>(0.0, 0.0, 0.0, 0.0);
            for (var xp: i32 = 0; xp < xmax; xp = xp + 1) {
                var pixel: vec4<f32>;
                if (sx + xp >= params.source_rect.x && sx + xp < params.source_rect.x + params.source_rect.z) {
                    pixel = read_input_at(vec2<i32>(sx + xp, sy + yp));
                    pixel = draw_pixel(pixel, u32(sx + xp), u32(sy + yp), true);
                    if (fix_range) {
                        pixel = remap_colorrange(pixel, bytes_per_pixel == 1);
                    }
                } else {
                    pixel = bg;
                }
                xsum = xsum + (pixel * kx[xp]);
            }
            sum = sum + xsum * ky[yp];
        } else {
            sum = sum + bg * ky[yp];
        }
    }
    return vec4<f32>(
        min(sum.x, params.pixel_value_limit),
        min(sum.y, params.pixel_value_limit),
        min(sum.z, params.pixel_value_limit),
        min(sum.w, params.pixel_value_limit)
    );
}
```
Replaces `sample_input_at`

VladimirP1, 2024-05-05:
I tried implementing the same algorithm as is used in imagemagick's distort operator - elliptical weighted average with cubic bc filtering. This is just a test to evaluate performance, it does not do any real distortion, it only applies a given affine transformation https://github.com/VladimirP1/gpu-warp . To use this in gyroflow we'd have to calculate affine approximations of the transformation at each pixel of the undistorted image, which is not much harder than computing the transformation itself.

Some test transformation 4000x3000 -> 1920x1080 + some rotation:
https://drive.google.com/drive/folders/1jHVp6L73TESmmYO1VOXEmn1dAzxfqL-H?usp=sharing
This exact transformation takes 54ms on UHD 630 and 20ms on GTX1070

VladimirP1, 2024-05-07:
I seem to have fixed most bugs and tried some real warping with my code

This shows that both upscale (in the left part of image) and downscale (in the right) work
[image]

This is just some fisheye-like distortion
[image]

temp-64GTX, 2024-05-08:
> tried some real warping

hmm. looks interesting. Does it possible to implement some custom warping? Like, you now, lens distortion:
https://github.com/gyroflow/gyroflow/issues/355

AdrianEddy, 2024-05-08:
> > tried some real warping
> 
> hmm. looks interesting. Does it possible to implement some custom warping? Like, you now, lens distortion: #355

that's not related, it's a different issue

VladimirP1, 2024-05-09:
Progress so far:
EWA, cubic bc filtering: https://youtu.be/egGW8EQafhc
current filtering in gyroflow (lanczos4): https://youtu.be/KNUOr-IasBg

I am using numeric differentiation now (so basically running the distort transformation three times per output pixel instead of one). It is possible to do it in one step, but that would require adding jacobian calculation into lens models.

AdrianEddy, 2024-10-06:
try latest dev build. I added a few interpolation methods based on Elliptical Weighted Average. The `RobidouxSharp` looks best in my option and I'm thinking to make it the default. Please experiment with different files and output sizes and let me know

If you want to build from source, you can even experiment with other B and C coefficients, put them [here](https://github.com/gyroflow/gyroflow/blob/511a8bc176d849d8121e78faf4b0fe9f3327a9f7/src/core/stabilization/mod.rs#L230-L233) from [here](https://github.com/ImageMagick/ImageMagick/blob/0ca37b0de25df22bdd4722a97368a089fef1eeb8/MagickCore/resize.c#L858-L893)

temp-64GTX, 2024-10-07:
Well, it's definitely working. As for the best method - i will compare them later today. 
But i guess it's on CPU now? Rendering process is 6 times slower on new version. Compared to version from 12.07.2024

[image]
[image]

AdrianEddy, 2024-10-07:
no, it's on GPU and should be even faster. What's in the log?

temp-64GTX, 2024-10-07:
Here is the log: 

[log.log](https://github.com/user-attachments/files/17275945/log.log)

AdrianEddy, 2024-10-07:
hmm, log looks ok. What's the rendering speed with Lanczos4? and what's the rendering speed in the previous build (gh2150)
https://github.com/gyroflow/gyroflow/actions/runs/11202849808

temp-64GTX, 2024-10-07:
Lanczos4 - speed is also slow (23 fps).
gh2150 - speed is also slow. 
I assume there is something else affecting it. I can try to step-by-step go to the older builds.

temp-64GTX, 2024-10-07:
Ok, problem starts here:
https://github.com/gyroflow/gyroflow/actions/runs/10575755096
_Update dependencies Release #2080: Commit [949ffd2]_

AdrianEddy, 2024-10-07:
Interesting, so just to be sure, gh2079 but 2080 is slow? Can you please check if there are differences between OpenCL and wgpu device under "Device for video processing"?

Also, what's your exact hardware?

temp-64GTX, 2024-10-07:
That's right, gh2079 is fast, 2080 is slow.

Results is odd. (gh2151)
OpenCL 3090Ti **77 fps**
OpenCL 4090 **24 fps**
wgpu Vulcan 4090 **22 fps**
wgpu Vulcan 3090Ti **82 fps**
wgpu DX12 4090 **80 fps**
wgpu DX12 3090Ti **78 fps**
wgpu GI 4090 **79 fps**
CPU only **25 fps**
gh2079 (Vulcan 4090) - 132 fps
gh2078 (Vulcan 4090) - 144 fps

I also noticed that the GPU core speed is stays at 210 Mhz on slow versions. And goes to 2860 Mhz on fast versions. 

My hardware:
Ryzen 9 5950X (OC 4.6 Ghz); DDR4 128Gb G Skill TridentZ Royal ; RTX 3090 Ti (Main for monitor); RTX 4090 (for render); Asus ROG Crosshair VIII Dark Hero; SSD Samsung 970 PRO 512 Gb; HDD 14Tb; Win10 x64 LTSC 21H2.

AdrianEddy, 2024-10-07:
What I suspect is that for some reason, there is an exchange of data between 3090 Ti and 4090 which makes it slow, whereas in the fast versions, everything happens on one GPU. Can you try to check what happens if you change "Device for rendering" (which is under Export settings->Advanced)?

temp-64GTX, 2024-10-07:
Double-odd.

[image]

closed 2024-10-07 by AdrianEddy