gyroflow/gyroflow #781: A simple stabilization method for damaged gyro data.

rainbowromka, 2024-01-08:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

The latest version has added a very simple way to stabilize files without gyroscope data. Suppose there is a video with damaged gyroscope data. Make it so that you can load it without gyroscope data and stabilize it in a simple way. It might be worth making sure that part of the video is stabilized by gyroscope data, and the other part in a simple way.

AdrianEddy, 2024-01-08:
Do you have a sample file?

rainbowromka, 2024-01-09:
Yes, I have. During the day I will try to upload it somewhere to the cloud. But for example, there is a problem with the gyro noise of the O3 air unit camera.

AdrianEddy, 2024-01-09:
In case of O3 noise it's much better to fix the hardware than to rely on optical flow, which will inherently be prone to errors. Usually just changing PWM to 48 kHz fixes the issue https://docs.gyroflow.xyz/app/getting-started/supported-cameras/dji#dji-o3-air-unit-vibration-issues

rainbowromka, 2024-01-09:
I only offered a tool. That case when the gyroscope fails you, you save the situation optically. O3 is given here as an example. For example, there are cases when the GoPro6 camera does not save gyroscope data or partially saves it. I shoot a lot in the mountains in the cold. Equipment often breaks down. I just love that you added a tool like this. But there is no universality. Before discarding valuable material, you can try to extract the maximum quality from it using professional tools.

AdrianEddy, 2024-01-18:
ok so this is already tracked in #45
the reason it's not closed is because it is planned to be much better and will actually have all the things you're asking.
The addition in v1.5.4 was just the simplest first step, but it's nowhere near completed, so you can watch #45 to get notified when more features related to it are added

closed 2024-01-18 by AdrianEddy