gyroflow/gyroflow #785: Gyroflow plugin causes Davinci Resolve to crash every time I open a project file containing it

hiktak, 2024-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.4

### What operating system are you using?

MacOS Sonoma 14.2.1

### What GPU are you using?

Radeon Pro 560X 4GB 

### What happened?

I had serious problems using Gyroflow plugins on BRAW footage, as soon as I added the Gyroflow plugins to the footage, my Mac's screen **went red** for 0.5 seconds and **Davinci Resolve crashed immediately**. after that. When I tried to restart Davinci Resolve and reopen the project file as above (which had been autosaved before), Davinci continued to crash again. When I tried creating another project file that didn't use gyroflow in it, Davinci Resolve still worked normally and never crashed like that.

I've tried customizing the GPU settings in Davinci Resolve, from Metal to OpenCPL, but the problem remains the same.

### Relevant log output

```shell
https://drive.google.com/file/d/1eaYwbGBh0VoeeXJaol6om1ZnAV-L9PNC/view?usp=sharing
```

AdrianEddy, 2024-01-13:
Did you try latest dev build of the plugin? https://gyroflow.xyz/devbuild/

hiktak, 2024-01-13:
> Did you try latest dev build of the plugin? https://gyroflow.xyz/devbuild/

Yes, I have tried this nightly build version, nothing's changed

AdrianEddy, 2024-01-13:
What's the resolution of your BRAW and timeline? Can you try setting the timeline resolution to 1080p or 2160p and then add the plugin?

hiktak, 2024-01-17:
> What's the resolution of your BRAW and timeline? Can you try setting the timeline resolution to 1080p or 2160p and then add the plugin?

I tried changing timeline resolution to 1080p today with OpenCL and it seemed to work, but sometimes it crashes sometimes not, and especially crashes often happen when adjusting the Colors Tab. As for Metal, this method is completely ineffective, I have no way to make it work with Metal. 

But when used with OpenCL, another problem occurs, when I load the project file with the Gyroflow plugin (which has been stabilized previously in the Gyroflow software), nothing seems to change in Davinci Resolve, I mean in Davinci Resolve the footage is not stabilized at all, even though the project file has been loaded. I also tried to reload the project file using Gyroflow software, I received this error message:

**"An error occurs: Filesystem error IOError(Os {code:1, kind: Permission Denied, message: "Operation not permitted"})**

AdrianEddy, 2024-01-18:
Please try latest dev build of both plugin and the app: https://gyroflow.xyz/devbuild/

It should work with Metal

What's the resolution of your BRAW?
What is your computer model/specs?

hiktak, 2024-01-21:
> Please try latest dev build of both plugin and the app: https://gyroflow.xyz/devbuild/
> 
> It should work with Metal
> 
> What's the resolution of your BRAW? What is your computer model/specs?

Following is my temporary way and somehow it worked:

- Install the lastest dev app & plugin as you said
- GPU settings in Davinci Resolve changed from Metal to OpenCL
- Change timeline resolution to 1080p, proxy preview resolution from Full to Half
- At the Render tab, turn off live preview during the rendering process

But I think the plugin needs to be optimized in future versions to work well with Metal, and should not crash anymore. I think the problem here is probably not in the hardware because when removing the Gyroflow plugin from the clip, my Intel iMac can easily handle 4K color grading and editing very smooth without any problems.

Once again thank you for your help.

AdrianEddy, 2024-01-21:
It is already optimized to work with Metal, that's why I'm trying to figure out your setup why it's not working for your case

What's the resolution of your BRAW? What is your computer model/specs?

hiktak, 2024-01-21:
> It is already optimized to work with Metal, that's why I'm trying to figure out your setup why it's not working for your case
> 
> What's the resolution of your BRAW? What is your computer model/specs?

Thank you for quick reply!

- BRAW Resolution is 4K DCI
- Computer model: iMac 21.5 2020 Retina 4K/ 3.2GHz 6-core 8th-generation Intel Core i7 processor, Turbo Boost up to 4.6GHz/ RAM 16GB DDR4 Onboard memory/ 512GB SSD storage/ GPU Radeon Pro 560X 4GB