gyroflow/gyroflow #786: Gyroflow as a full movie edition

eirasys, 2024-01-17:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Gyroflow is one of the best because it supports a lot of video format, frame rates and also gyro data.
The feature request concept is simple, use Gyroflow to edit and mount all the drone/camera videos:

1. Possibility to create an array sequence of trims [index i, trim start i, trim end I] and export them as a single video. You can even overlap trims to repeat an important sequence (for example use several colors for each trim). Each trim should have its own export settings preset. Edition could be made by selecting a trim index and adding or deleting indexes.
2. Enable an easier way to join videos after having one opened, not only with the "drop video file" in the beginning. Include the gyro data is very important when adding those files, and also checking compatibility between them (same lens, same gyro settings)
3. When several videos are joined, we should be able to export them with the camera information included. For example we may want to join several movies from the drone, and process its camera correction and stabilization only in the end!? Not sure about this one, may be solved with request 1.
4. Allow color correction and color profiles to be processed and exported.

Thanks for this excellent video tool,
EiraSYS

AdrianEddy, 2024-01-17:
This is out of scope for Gyroflow. 

While it would be cool, it's an insane amount of work for a one-person project, not mentioning that there are still insane amount of tasks that are **in** scope, so unless someone joins the project, it's not going to happen due to limited manpower. 

I'm working on Adobe plugin right now, so Gyroflow will be available for all common video editors as a plugin, which is a much better option than building a video editor inside Gyroflow

That being said, number 1. was just added and number 4. will be done with LUT support (https://github.com/gyroflow/gyroflow/issues/250)

closed 2024-01-17 by AdrianEddy

eirasys, 2024-01-21:
I understand! I'm working with Final Cut Pro on Mac, do you plan any plugin for it?
I guess that the array of streams would be quite easy todo. The problem with other editors is that they reprocess all the video. I don't know how you do it, but when you bind several video files of the drone the result is a long video that is the sum in bytes  of the other ones. It also has better quality, faster and you store all gyro data. Perfect! Using other tools I can't get the same result, always much processing time, lower quality and worse, the resulting file occupies a lot of space in the disk...

AdrianEddy, 2024-01-21:
FCPX already has a plugin https://gyroflowtoolbox.io/

If you're working with files from the same camera and settings, you can merge them all, then use multiple trims to export only the needed parts, which will be somewhat similar to a full video editor