gyroflow/gyroflow #787: Video quality when working with BRAW files

hiktak, 2024-01-21:
I realized that video stabilized with the Gyroflow plugin and native Davinci Resolve's gyro stabilization **have different quality**. The video stabilized with Gyroflow have less detail and sharpness. Please see a visual example here, especially when you look at the flowers and trees:  [](https://drive.google.com/drive/folders/1AdgHOs2ERAbEgxHBn2o2ddjfWDmbOqrm?usp=sharing)

I'm working on BRAW 4K DCI, no sharpen added and export settings are the same in both files. As for as I know Gyroflow only affects gyro data and all exporting is done in Davinci Resolve, but it seems like the Gyroflow algorithm added motion blur or reduced bitrate I don't know? 

Thank you for your help.

AdrianEddy, 2024-01-21:
Hmm Gyroflow plugin is not doing the rendering so it can't affect bitrate and it definitely doesn't add motion blur. Are you sure you're exporting 4k? In the other issue you mentioned changing the resolution to 1080p

hiktak, 2024-01-21:
> Hmm Gyroflow plugin is not doing the rendering so it can't affect bitrate and it definitely doesn't add motion blur. Are you sure you're exporting 4k? In the other issue you mentioned changing the resolution to 1080p

Yes, I'm sure it's 4K because before rendering I switched the timeline resolution back to 4K, 1080p only when editing to avoid crashing. I have attached 2 videos in the link above

AdrianEddy, 2024-01-21:
Can you also upload the original BRAW file?

hiktak, 2024-01-21:
> Can you also upload the original BRAW file?

Sure! Here it is: https://drive.google.com/drive/folders/1EqnV11JfcuQ6PT4TdT0XoeJfda735iBQ?usp=share_link

closed 2024-02-05 by AdrianEddy