gyroflow/gyroflow #788: Change Zooming center offset for different parts of a video

Leseratte10, 2024-01-21:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I'm trying to stabilize a video using dynamic zoom, but I don't always like the way it zooms - no matter what offset I use, in some parts of the video the subject is too far to the left (cut off) and in some parts it's too far to the right. 

Is there a way (or if there isn't, can this feature be added to GyroFlow) to allow me to change the zoom center offset (or, in general, most the stabilization settings) for different sections of the video?

I'd like a setting like "from 0:00 to 2:00 use X offset of -6%, then gradually fade to using an X offset of -7% from 2:05 to 4:00" or something like that. That way it'd use an offset of -6% for the first two minutes of the video, and then over a span of 5 seconds it'd change the offset to 7% and then keep it there.

Not sure what I'm doing wrong. Static zoom doesn't work because it cuts off part of the content at the left and right sides just because there's parts in the video (in the beginning and end) where that would be needed. Dynamic zoom kinda works, but obviously doesn't "know" which part of the screen is the actual subject that should be in focus, and I can only change that (with the zooming center offset) for an entire video. 

I checked the documentation and didn't find anything related to that. Am I missing something and/or using Gyroflow wrong or is this not supported yet? 

Ideally, I would like to have it in a way where I could change every single setting in the right pane of Gyroflow (under "Syncronization" and "Stabilization") either for the whole video (as it is now) or just for parts of one video, with some kind of smoothing when the setting changes so there's no hard cut.

AdrianEddy, 2024-01-21:
Yes it's called [keyframing](https://docs.gyroflow.xyz/app/advanced-usage/keyframes)

closed 2024-01-21 by AdrianEddy

Leseratte10, 2024-01-21:
Oh wow, that was a quick response. Thanks. I only knew keyframes as part of video encoding so I skipped over that part of the docs, and I looked in the context menu for the timeline and didn't even know the seperate sliders had context menus. 

This seems to be exactly what I need - great!