gyroflow/gyroflow #790: Ability to export 3D "solved" camera Data

Joegenco, 2024-01-26:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hello! I've been looking to build a camera tracking workflow which uses the gyro data to get accurate camera orientation.

I stumbled upon gyroflow and couldn't find any references to exporting a 3D camera from it.

Is there a way to export the resulting camera track as an FBX or at-least a Json file to bring into a 3d software to reconstruct a 3D camera solve?

This would be really helpful, your time responding is much appreciated!

AdrianEddy, 2024-01-26:
There's https://github.com/EmberLightVFX/Gyroflow-to-CSV
but with version v1.5.1 or later, use this updated script: https://github.com/EmberLightVFX/Gyroflow-to-CSV/issues/6#issuecomment-1773736663

jumpjack, 2024-03-01:
Does anybody know how to use such exported data to feed FFMPEG (or anything else) to stabilize a 360° equirectangular video?

AdrianEddy, 2024-06-16:
Latest dev build has a function to export gyro data to CSV or JSON.

Please test and let me know if it works for your use case.

Can someone prepare me a test file with a USD camera export using this data?

AdrianEddy, 2024-06-25:
I also added .USD and .JSX export

closed 2024-06-25 by AdrianEddy