gyroflow/gyroflow #791: Wrong aspect ratio/stretch in preview when Preserve export settings is checked (source 4x3, output 16x9)

dany123, 2024-01-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4 (same in v1.5.4 (gh1752) dev build)

### What operating system are you using?

Windows 10 x64

### What GPU are you using?

Nvidia 4070

### What happened?

I noticed a minor bug: when drag&dropping a second .mp4 file over the application, the preview window shows a 4x3 image stretched to a 16x9 area. This affects only preview image, not export. Can be fixed by selecting any 16x9 preset in the export panel.

Detalied steps:
1. open gyroflow and load any file (it seems the first file is properly shown)
2. go to the Export settings, choose 1920x1080 preset, check the "Preserve export settings" option
3. drag&drop a .mp4 file with 4x3 aspect ratio (in my case I have 2704x2028 60fps and 1920x1440 60fps videos from a gopro7)
4. look at the shown preview, video is stabilized properly (auto sync enabled), but the video preview is a stretched 4x3 -> 16x9 view (see screenshot)
5. to fix it: go to the Export settings and click on any preset, it will properly apply crop instead of stretch
(as I mentioned, this is only affecting preview, export seems to be properly cropped instead of stretched)

before & after preset reset:
[image]
[image]

AdrianEddy, 2024-02-02:
hmm, I can't reproduce this with my files, could you upload your sample files?

dany123, 2024-02-02:
Here's a quick sample video https://we.tl/t-G7ops6m7Ma I reproduced this both on Windows 10 PC and on my Windows 11 laptop, standalone .zip latest Gyroflow freshly downloaded.

Just load the file, (select lens: Gopro 7 4:3 wide NO-EIS - the official one, for some reason it didn't auto load for this file), go to export settings, select 1920x1080 then in Advanced check Preserve export settings.

The 2nd time I drag and drop the file it is shown stretched. Re-clicking the 1920x1080 preset to fix it makes this very obvious. Dropping the file again stretches it again (after doing the auto sync).
Note in the recording the blue part is a 16x9 area, the red top/bottom make up 4x3 and the side black areas make up the entire monitor at 21x9 - I was hoping it'd help but I should have used a checkered pattern or something. Anyway, let me know if you reproduce it with this or you need a different sample.

AdrianEddy, 2024-02-05:
Fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2024-02-05 by AdrianEddy