gyroflow/gyroflow #792: Gyroflow crashing on iPad when stabilising without gyro data (optical flow)

DronePowerWales, 2024-02-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4.1745

### What operating system are you using?

IOS 16.5.1 (20F75)

### What happened?

[My_project_25.zip](https://github.com/gyroflow/gyroflow/files/14200239/My_project_25.zip)
Hi, I am experiencing an issue with gyroflow, when trying to stabilise footage without gyro data on my iPad the app crashes around 50%. I was wondering if this could be fixed? Thanks

DronePowerWales, 2024-02-07:
This could be because of the iPad running out of ram to store new frames which have been stabilised? I am using a iPad air 4.

DronePowerWales, 2024-02-07:
This could be because the iPads ram in becoming full of

DronePowerWales, 2024-02-07:
This could be because the iPads ram is becoming full of stabilised frames therefore causing Gyroflow to crash? I am using a iPad air 4.

AdrianEddy, 2024-05-23:
Please try latest build from TestFlight https://testflight.apple.com/join/e0RI1n5i

AdrianEddy, 2024-05-25:
This should be fixed in the latest TestFlight build

closed 2024-05-25 by AdrianEddy

DronePowerWales, 2024-05-25:
> stabilising without gyro data (optical flow) (Issue #792)

Thank you for fixing this issue, when testing the latest build on test 
flight this issue is fixed however it is crashing when I attempt to 
export the stabilised video and I was wondering if you would be able to 
fix this?

  Thanks for your help

Sent via BT Email App

AdrianEddy, 2024-05-25:
The crash on export should also be fixed in latest one (updated today) version 1.54.1913. have you tried this one?

DronePowerWales, 2024-05-29:
Hi Adrian, sorry for the late reply, I have just tested the latest 
version of gyroflow on testflight and everything seems to be working 
fine with no crashes. Thank you for all your help 👍