gyroflow/gyroflow #793: Parse accelerometer metadata from mp4 files recorded on Android

UnlimitedIllusionsSoftware, 2024-02-10:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

If I understand correctly accelerometer metadata is needed to make "Horizon Lock" feature work.
Is that possible to add support for such kind of metadata similar with https://github.com/gyroflow/gyroflow/issues/753?
 (big-endian, "mett" box, application/gyro and application/accel mime type)

AdrianEddy, 2024-02-11:
Can you share a sample file?

UnlimitedIllusionsSoftware, 2024-02-12:
Sure, here it is
https://drive.google.com/file/d/1smWmTq_zUF5AE8Ut_keDiUKNhOGgFiZq/view?usp=sharing

AdrianEddy, 2024-02-13:
That file seems corrupted, the `stbl` for the accel track is empty:
[image]

UnlimitedIllusionsSoftware, 2024-02-14:
Could you please try this one:
https://drive.google.com/file/d/1L7GJsNqf8JX6wsYDCNfsx8qMNLYoUhSh/view?usp=sharing

AdrianEddy, 2024-02-14:
this one is good. What's the unit of the accelerometer? `m/s²` or `g`?

AdrianEddy, 2024-02-14:
Done in https://github.com/AdrianEddy/telemetry-parser/commit/3097e1b6c2750256f4c021e725cc822d58e5d1ea
Latest dev build: https://gyroflow.xyz/devbuild/

closed 2024-02-14 by AdrianEddy