gyroflow/gyroflow #794: Gyroflow 1.5.4 in VMWare Debian 12 (no GUI)

salem83, 2024-02-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

Gyroflow 1.5.4 (gh1756)

### What operating system are you using?

Debian 12 

### What GPU are you using?

VMWware SVGA II Adapter

### What happened?

Gyroflow will quit with the following error messages

### Relevant log output

```shell
19:31:28 [INFO] [MDK] 0.25.0 (git 26e18d1) - Multimedia Development Kit. Copyright (c) 2016-2024 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang18.1.0 (++20240203081317+140ad18e1ea2-1~exp1~20240203201335.24)GCC4.2.1; libc++180100; 08:40:16 Feb  7 2024
19:31:28 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
19:31:28 [DEBUG] (1) gyroflow: Gyroflow 1.5.4 (gh1756)
19:31:28 [INFO] Videos: ["../GX010262.MP4"]
19:31:28 [INFO] Presets: ["../TestExport.gyroflow"]
19:31:28 [INFO] Lens profiles directory: "/home/mrt/Gyroflow-dev/camera_presets"
19:31:28 [INFO] Loaded 10119 lens profiles in 307.745ms
19:31:38 [ERROR] thread 'main' panicked at 'Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))': /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ocl-0.19.6/src/standard/platform.rs:49
   0: <unknown>
   1: <unknown>
   2: <unknown>
   3: <unknown>
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: __libc_start_main
  16: <unknown>

19:31:38 [ERROR] Failed to initialize OpenCL Platform::list: Error retrieving platform list: ApiWrapper(GetPlatformIdsPlatformListUnavailable(10))
19:31:39 [ERROR] wgpu init error
19:31:39 [ERROR] NotAFile("../GX010262.MP4")
19:31:39 [WARN] [add_file]: Unable to read the video file.
19:31:39 [ERROR] [ef83aa7d] Error: An error occured: Unable to read the video file.

with NO_OPENCL=1
0.25.0 (git 26e18d1) - Multimedia Development Kit. Copyright (c) 2016-2024 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang18.1.0 (++20240203081317+140ad18e1ea2-1~exp1~20240203201335.24)GCC4.2.1; libc++180100; 08:40:16 Feb  7 2024
19:44:10 [DEBUG] (1) gyroflow: [in main] Qt version: 6.4.3
19:44:10 [DEBUG] (1) gyroflow: Gyroflow 1.5.4 (gh1756)
19:44:10 [INFO] Videos: ["../GX010262.MP4"]
19:44:10 [INFO] Presets: ["../TestExport.gyroflow"]
19:44:10 [INFO] Lens profiles directory: "/home/mrt/Gyroflow-dev/camera_presets"
19:44:10 [INFO] Loaded 10119 lens profiles in 108.454ms
19:44:10 [ERROR] wgpu init error
[src/cli.rs:418:5] &settings = {}
19:44:10 [ERROR] NotAFile("../GX010262.MP4")
19:44:10 [WARN] [add_file]: Unable to read the video file.
19:44:10 [ERROR] [8b7ea298] Error: An error occured: Unable to read the video file.
```

AdrianEddy, 2024-02-11:
it says "Not a file" are you sure you're entering the correct path to file?

salem83, 2024-02-11:
Well, the paths for the files need to be absolute. That was the issue. 

What about the OpenCL error?
This issue has already been reported here #672. Is there any suitable package for Debian?

Thanks

AdrianEddy, 2024-02-11:
VMware virtual machines don't have a GPU, so you won't get a working OpenCL there. You can just ignore the error or set NO_OPENCL=1 environment variable to skip trying to initialize OpenCL

closed 2024-02-11 by salem83