gyroflow/gyroflow #795: Gyroflow causing total pc crash (blue screen) when loading video

aczubak, 2024-02-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4 and newest dev (Gh1756)

### What operating system are you using?

windows 11 home 64-bit (10.0,build 22621)

### What GPU are you using?

Intel arc A370M

### What happened?

So the program was working well last few weeks.  However today when loading videos it started blue screening my laptop.

The only thing that changed is that i did update my video card drivers this morning. So i'm guessing that could have something to do with it?  Video driver:  31.0.101.5186

Not sure what else would have changed.   I'm unable to grab any log from the console as the whole PC blue screens.   is there some log i can grab anywhere else to help?

Thanks,
--Adam

AdrianEddy, 2024-02-13:
GPU drivers are at fault here, try to install some different ones. I don't think there's anything I can do on my side, it's definitely some driver bug
You can try to run with NO_OPENCL=1 environment variable set to disable OpenCL processing

aczubak, 2024-02-14:
You were right.  its weird i did an "official" update through HP and it
broke.  I tried the Open cl, still was broke.  then went to Intel and did a
clean install from them.  solved the problem.  So moving forward will
utilize Intel drivers directly.

Appreciate the response!

Thanks,
--Adam

closed 2024-02-14 by AdrianEddy