gyroflow/gyroflow #797: Update creation date metadata time when trimming the start of the clip

dany123, 2024-02-17:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I usually export using gyroflow smaller portions of the original file and edit these in another app. When doing this, the new file has the same identical creation time* as the original file, which messes up a GPS** overlay I'm using (which basically looks at the create date metadata and does the extra sync work automatically).

So a quick update of this time would help a lot and would technically be more correct. Thanks.

*I am referring to the creation_time metadata, not file creation/modified data - I do not care about these altough it would make sense to update them too
**note that I'm using an external .gpx file, so this isn't related to the embedded gps tracks timing

AdrianEddy, 2024-02-17:
Done in latest dev build https://gyroflow.xyz/devbuild/

closed 2024-02-17 by AdrianEddy

dany123, 2024-02-17:
Did a quick test with the dev build and works perfectly. Thank you!