gyroflow/gyroflow #803: When opening the 4K10bit4:2:2 video captured by A7R5 multiple times, the gyroflow crashes

wzy-icecream, 2024-03-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.4

### What operating system are you using?

windows11 

### What GPU are you using?

4070ti

### What happened?

[229afbf0-cd13-406b-bdd4-d0b01993c800.dmp](https://github.com/gyroflow/gyroflow/files/14550930/229afbf0-cd13-406b-bdd4-d0b01993c800.dmp)
Using drag and drop to open files，After multiple attempts to open the file, a crash occurred. dmp file isalready upload
[229afbf0-cd13-406b-bdd4-d0b01993c800.dmp](https://github.com/gyroflow/gyroflow/files/14550934/229afbf0-cd13-406b-bdd4-d0b01993c800.dmp)

AdrianEddy, 2024-03-10:
Latest dev build? https://gyroflow.xyz/devbuild

wzy-icecream, 2024-03-10:
Hello,Eddy
No，I use the last release build， I just used the latest dev version and this issue has been fixed.
Thanks you

closed 2024-03-10 by AdrianEddy