gyroflow/gyroflow #804: Add support for ARRI cameras

grosbecyk, 2024-03-11:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hello,

First, thank you for the very nice software.

The ARRI 35 camera seems to have a built-in gyro recording Tilt and Roll Values which could be used for stabilization, plus other metadata tied to the lens. Some other ARRI models may or may not have the orientation data, I'm not sure.

ARRI metadata informations:
https://www.arri.com/en/learn-help/learn-help-camera-system/pre-postproduction/metadata

Sample files are supplied here (I would recommend ):
https://www.arri.com/en/learn-help/learn-help-camera-system/camera-sample-footage-reference-image

The ARRI Reference Tool can export .json data:
https://www.arri.com/en/learn-help/learn-help-camera-system/tools/arri-reference-tool

Example:
`"frames" : [
            {
                "frameId" : 0,
                "frameBasedMetadataSets" : {
                    "Filter" : {
                        "ndFilterDensity" : 1800,
                        "ndFilterIndex" : 3
                    },
                    "Lens State" : {
                        "lensFocalLength" : 32000,
                        "lensFocusDistanceImperial" : 160512,
                        "lensFocusDistanceMetric" : 4077,
                        "lensMotorRawEncoderValueFocus" : 2128
                    },
                    "Positional" : {
                        "orientation" : {
                            "tilt" : 178.8,
                            "roll" : 1.9
                        }
                    },
                    "Sensor State" : {
                        "averagedFrames" : 1,
                        "exposureIndex" : 800,
                        "exposureTime" : "20833/1000000",
                        "sensorSampleRate" : "0/0"
                    },
                    "White Balance" : {
                        "colorTemperature" : 5600,
                        "whiteBalanceTint" : 0
                    }
                }
            },`

Support for ARRI cameras would be VERY appreciated. Thanks

AdrianEddy, 2024-03-11:
Unfortunately tilt and roll per frame are not enough for stabilization, not even tilt/roll/yaw per frame are enough, the gyro needs to be sampled multiple times per frame in order to achieve any acceptable results. We're already tried cameras which have one value per frame and it never worked

grosbecyk, 2024-03-11:
Thank you for the answer.

closed 2024-03-11 by grosbecyk