gyroflow/gyroflow #805: How do you determine the timestamp of the first frame

RigerLee, 2024-03-15:
Sorry for bothering. I parse the metadata from telemetry-parser to get the timestamp of each frame, but in fact the total number of data given is more than the number of video frames. 

For example, if a video has a total of 10 frames, we may get the following:
```
Exposure                  Data                      Shutter speed                                     : [
TimeScalar { t: -0.3650000000000091, v: 0.019990403205156326 }, 
TimeScalar { t: -0.33100000000000307, v: 0.019990403205156326 }, 
TimeScalar { t: -0.2980000000000018, v: 0.019990403205156326 }, 
TimeScalar { t: -0.26500000000000057, v: 0.019990403205156326 }, 
TimeScalar { t: -0.23100000000000875, v: 0.019990403205156326 }, 
TimeScalar { t: -0.1980000000000075, v: 0.019990403205156326 }, 
TimeScalar { t: -0.16500000000000625, v: 0.019990403205156326 }, 
TimeScalar { t: -0.13100000000000023, v: 0.019990403205156326 }, 
TimeScalar { t: -0.09799999999999898, v: 0.019990403205156326 }, 
TimeScalar { t: -0.06400000000000716, v: 0.019990403205156326 }, 
TimeScalar { t: -0.03100000000000591, v: 0.019990403205156326 }, 
TimeScalar { t: 0.001999999999995339, v: 0.019990403205156326 }, 
TimeScalar { t: 0.036000000000001364, v: 0.019990403205156326 }, 
TimeScalar { t: 0.0689999999999884, v: 0.019990403205156326 }, 
TimeScalar { t: 0.10199999999998965, v: 0.019990403205156326 }, 
TimeScalar { t: 0.13599999999999568, v: 0.019990403205156326 }, 
TimeScalar { t: 0.16899999999999693, v: 0.019990403205156326 }, 
TimeScalar { t: 0.20199999999999818, v: 0.019990403205156326 }, 
TimeScalar { t: 0.23599999999999, v: 0.019990403205156326 }, 
TimeScalar { t: 0.26899999999999125, v: 0.019990403205156326 }, 
TimeScalar { t: 0.30299999999999727, v: 0.019990403205156326 }, 
TimeScalar { t: 0.3359999999999985, v: 0.019990403205156326 }, 
TimeScalar { t: 0.3689999999999998, v: 0.019990403205156326 }, 
TimeScalar { t: 0.4029999999999916, v: 0.019990403205156326 }, 
TimeScalar { t: 0.43599999999999284, v: 0.019990403205156326 }, 
```

How does gyroflow decide the timestamp of the first frame? 

Does it simply take the first item with t>= 0? Like `TimeScalar { t: 0.001999999999995339, v: 0.019990403205156326 }` in the above example.

AdrianEddy, 2024-03-15:
Every camera is different, but telemetry-parser tries to return `t = 0` at the middle of the first frame
Remember that there are many samples per frame, and the negative `t` is the top row of the first frame, then it goes to 0 in the middle and then positive at the bottom
That's assuming you're using  `util::normalized_imu()` or `util::normalized_imu_interpolated()`

Gyroflow also takes into account exposure time and some other data depending on camera, for Insta360 it's here:
https://github.com/gyroflow/gyroflow/blob/master/src/core/gyro_source.rs#L390-L414
This is important for the most accurate timing

RigerLee, 2024-03-15:
Thank you for your explanation. 

In the given code snippet, it seems that only Timescalers with t>=0 have been considered. 

Can I say that the first non negative Timescaler is the reference timestamp of the first frame, since the interval between two Timescaler is exactly 1/fps. 
And the actual timestamp of the first frame is obtained based on the formula in the code snippet.

AdrianEddy, 2024-03-15:
It's not that simple, every frame has an additional time offset, based on the exposure time, that offset is calculated in that code snippet I linked.
It also takes into account `gyro_timestamp` and `first_frame_timestamp`, but that's already calculated in `telemetry-parser`

To verify the timestamp per frame, you can Export project file `(with processed gyro data)` from Gyroflow, and then in that .gyroflow json file, you'll have `"synced_imu_timestamps_with_per_frame_offset"` field

RigerLee, 2024-03-19:
Ok, I'll try to use the synced imu timestamp to sync with another imu.

closed 2024-03-19 by RigerLee

XUTONG1024, 2024-07-04:
> Ok, I'll try to use the synced imu timestamp to sync with another imu.

hi,May I ask if you have made any progress on this issue? I am currently considering using the built-in IMU of Insta360 x2 to implement vio. Can you give me some suggestions?