gyroflow/gyroflow #807: Attempt at live video stabilization

ninono12345, 2024-03-18:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hello everyone,

I am a student of artificial intelligence. I am still in the learning phase, but already have some experience with difficult python and medium difficulty c++ code.
I have been assigned a project by my professor, find and implement an algorithm for stabilizing live video.
Even though there are other options than gyroflow, he suggested me this, because the stabilization quality is very good, and this can be done with live video, it is my task to find out how. And also to make it a real challenge, this is written in rust, which I know minimally, so at the same time this is an opportunity to learn a new language.

Now my equipment is a simple usb connected camera, and a mpu9250 connected through raspberry pi pico to my pc, outputting raw gyroscope and accelerometer data to the serial monitor at 100Hz. The idea is to glue the mpu9250 to the camera to receive the camera orientation.

I figure the first steps would be to get a lens profiling of the camera by filming a checkerboard and processing the footage through gyroflow app.
Second: record sample footage and at the same time collect gyroscope and accelerometer data, put it in the .gcsv file and then to try it out put video and data in gyroflow app to see If we are getting nice results.

Once we know that everything here is ok, now is the time to implement gyroflow_core

Because there is no documentation, I would have to read through and analyze the a lot of code to figure out what I need.
So far the understandable part is the imu_integration folder.
For each iteration that my mpu9250 gives out data I should use the VQFIntegrator (or perhaps some different) to calculate the quaternion coordinates?
Now for what to do next I am lost. Can somebody please give some pointers for what steps happen next for video stabilization?
Will stabilizing with cuda or cpu make the code differ a lot or just will it be the use of a different class file?
Also can somebody give me an insight at how does gyroflow work, that is are there held in a buffer, so current frame can be compared with the previous one? Or does a frame get "analyzed" and some specific data of the frame collected, then "thrown away"?

This is just an attempt of making a bare skeleton of live video stabilization, that is a simple usb connected camera, and on screen gets displayed a stabilized stream.

Any help is appreciated, thank you in advance :)

AdrianEddy, 2024-03-18:
Hi and welcome
Please join discord: https://discord.gg/WfxZZXjpke
There's a few points already discussed in the [#dev-live-stabilization](https://discord.com/channels/797044698682228736/1055194134622765077) channel, read the first message in that channel and then I'll be able to answer more of your questions there

ninono12345, 2024-03-18:
@AdrianEddy thank you for your quick response, I will look into it

ninono12345, 2024-03-18:
@AdrianEddy I wrote my questions in the discord chat

AdrianEddy, 2024-04-02:
Closing this as redundant, let's track it in #635

AdrianEddy, 2024-10-30:
https://discord.com/invite/BBJ2UVAr2D

joshuasv, 2024-10-30:
Unfortunately that invite has expired...

AdrianEddy, 2024-10-30:
https://discord.gg/YaUtNpWTUh

joshuasv, 2024-10-30:
Now it says it is invalid or that I may have not permission to join...

AdrianEddy, 2024-10-31:
Maybe something wrong with your Discord profile? maybe try to create a new one? The invite link is correct and active