gyroflow/gyroflow #808: Handling GPU drivers crash

czukowski, 2024-03-19:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

GPU drivers are crap, I guess... I've had some occasional random issues recently with black screen during playback or seeking on timeline. The OS (Windows here) recovers after a while but Gyroflow crashes as well.

Is it possible to handle it and dump the project file to prevent loss of progress?

The log file contains messages near the end: `Device loss detected in Present()`, `Graphics device lost, cleaning up scenegraph and releasing RHI` and then a lot of errors like `Failed to create buffer`, `Failed to build vertex/index buffer of size XXX`, `Failed to (re)build vertex/index buffer of size XXX` and `Failed to build uniform buffer of size XXX bytes`.

AdrianEddy, 2024-03-19:
was the driver crash caused by Gyroflow? Are you using BRAW or R3D? (they have their own GPU decoders in the SDK)
It is reproducible? Maybe try different "Preview pipeline"?

I'm not sure how to recover from that, there are too many moving parts here to catch, with the main one being Qt (and these error messages are from Qt)

czukowski, 2024-03-19:
I don't believe it was caused by Gyroflow as it's random (I can't see any apparent cause) and doesn't occur very often, but it's annoying because when it does, it means loss of some progress. Not using BRAW or R3D, will try a different pipeline.

AdrianEddy, 2024-05-10:
I'm closing this as I don't see any sensible way to handle this. There are too many moving parts and the GPU code is already too complex. I have way too much trouble keeping track of everything happening with the GPU and making it maintainable, that thinking about gracefully handling GPU crashes is just beyond my abilities and I'm unable to take this on my shoulders

closed 2024-05-10 by AdrianEddy