gyroflow/gyroflow #812: Final Cut Plugin issue (A plug-in is not responding)

yarovikov, 2024-03-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.1.4

### What operating system are you using?

MacOS Sonoma 14.3 (23D56)

### What happened?

Hi there
I have 
- Macbook (2,6 GHz 6-Core Intel Core i7, AMD Radeon Pro 5300M 4 GB Intel UHD Graphics 630 1536 MB), 
- MacOS Sonoma 14.3 (23D56), 
- Final Cut X 10.7, 
- Installed Gyroflow Toolbox 1.1.4.

This happens when i use import gyroflow project file (with procceded gyro data) in the plugin options. 

[image]

Any ideas? 
Thanks!

AdrianEddy, 2024-03-31:
@latenitefilms

latenitefilms, 2024-03-31:
Sorry you're having issues @yarovikov !

I'm away from a computer currently, but to diagnose further can you please supply any crash logs? You can find them in `~/Library/Logs/DiagnosticReports`.

We'll also need your log files, which I THINK are in `~/Library/Group Containers/A5HDJTY9X5.com.latenitefilms.GyroflowToolbox/Library/Application Support/`. I'll triple check when back in front of a computer.

yarovikov, 2024-04-28:
hi. I apologize for the long response

<details>
  <summary>crash log</summary>

  Date/Time:        2024-04-28 16:29:59.842 +0100
End time:         2024-04-28 16:32:35.574 +0100
OS Version:       macOS 14.3 (Build 23D56)
Architecture:     x86_64h
Report Version:   44
Incident Identifier: 5B5FB936-9D36-416D-851E-F7CF7E45116D

Data Source:      Microstackshots
Shared Cache:     D956D414-4FAA-3DF6-AE76-21902DB97156 slid base address 0x7ff803218000, slide 0x3218000

Command:          Gyroflow Toolbox Renderer
Path:             /Applications/Gyroflow Toolbox.app/Contents/PlugIns/Gyroflow Toolbox Renderer.pluginkit/Contents/MacOS/Gyroflow Toolbox Renderer
Identifier:       com.latenitefilms.GyroflowToolbox.Renderer
Version:          1.1.4 (18)
Team ID:          A5HDJTY9X5
Resource Coalition ID: 26178
On Behalf Of:     3 samples UNKNOWN [40261]
Architecture:     x86_64
PID:              45801

Event:            wakeups
Action taken:     none
Wakeups:          45001 wakeups over the last 156 seconds (289 wakeups per second average), exceeding limit of 150 wakeups per second over 300 seconds
Wakeups limit:    45000
Limit duration:   300s
Wakeups caused:   45001
Wakeups duration: 156s
Duration:         155.73s
Duration Sampled: 134.32s
Steps:            4

Hardware model:   MacBookPro16,1
Active cpus:      12
HW page size:     4096
VM page size:     4096

Fan speed:        5298 rpm
OS Cryptex File Extents: 106

Heaviest stack for the target process:
  3  start_wqthread + 15 (libsystem_pthread.dylib + 7063) [0x7ff80364eb97]
  3  _pthread_wqthread + 327 (libsystem_pthread.dylib + 11335) [0x7ff80364fc47]
  3  _dispatch_workloop_worker_thread + 451 (libdispatch.dylib + 78845) [0x7ff8034bb3fd]
  3  _dispatch_root_queue_drain_deferred_wlh + 271 (libdispatch.dylib + 80622) [0x7ff8034bbaee]
  3  _dispatch_lane_invoke + 377 (libdispatch.dylib + 41216) [0x7ff8034b2100]
  3  _dispatch_lane_serial_drain + 387 (libdispatch.dylib + 38075) [0x7ff8034b14bb]
  3  _dispatch_mach_invoke + 497 (libdispatch.dylib + 117877) [0x7ff8034c4c75]
  3  _dispatch_lane_serial_drain + 387 (libdispatch.dylib + 38075) [0x7ff8034b14bb]
  3  _dispatch_mach_msg_invoke + 450 (libdispatch.dylib + 115030) [0x7ff8034c4156]
  3  _dispatch_client_callout4 + 9 (libdispatch.dylib + 13876) [0x7ff8034ab634]
  3  _xpc_connection_mach_event + 1366 (libxpc.dylib + 49455) [0x7ff80336f12f]
  3  _xpc_connection_call_event_handler + 56 (libxpc.dylib + 54360) [0x7ff803370458]
  3  message_handler + 140 (Foundation + 9129928) [0x7ff804f20fc8]
  3  message_handler_message + 80 (Foundation + 9131180) [0x7ff804f214ac]
  2  -[NSXPCConnection _decodeAndInvokeMessageWithEvent:reply:flags:] + 1661 (Foundation + 9124475) [0x7ff804f1fa7b]
  2  __NSXPCCONNECTION_IS_CALLING_OUT_TO_EXPORTED_OBJECT__ + 17 (Foundation + 732684) [0x7ff80471ee0c]
  2  -[NSInvocation invoke] + 302 (CoreFoundation + 383334) [0x7ff803710966]
  2  __invoking___ + 140 (CoreFoundation + 383676) [0x7ff803710abc]
  1  -[FxRemotePluginCoordinator restartSession:forPlugInWithClass:pluginUUID:version:isInDocument:reply:] + 114 (FxPlug + 222516) [0x10a238534]
  1  createPlugin(NSString*, NSString*, FxRemotePluginCoordinator*, unsigned int, unsigned long, NSMutableDictionary<NSNumber*, FxRemotePlugin*>*, NSError**) + 215 (FxPlug + 220965) [0x10a237f25]
  1  ??? (Gyroflow Toolbox Renderer + 18592) [0x109a538a0]
  1  ??? (Gyroflow Toolbox Renderer + 61687) [0x109a5e0f7]
  1  +[NSData(NSData) dataWithContentsOfFile:] + 33 (Foundation + 534470) [0x7ff8046ee7c6]
  1  -[NSData(NSData) initWithContentsOfFile:] + 92 (Foundation + 218729) [0x7ff8046a1669]
  1  _NSReadBytesFromFileWithExtendedAttributes + 167 (Foundation + 145576) [0x7ff80468f8a8]
[... 66 lines cut ...]
                                                    1  _decodeObject + 1259 (Foundation + 9158314) [0x7ff804f27eaa]
                                                      1  -[FxStringParameterTransaction initWithCoder:] + 131 (FxPlug + 123949) [0x10a22042d]
                                                        1  -[NSXPCDecoder decodeObjectOfClasses:forKey:] + 38 (Foundation + 230372) [0x7ff8046a43e4]
                                                          1  _decodeObjectAfterSettingAllowListForKey + 144 (Foundation + 9159360) [0x7ff804f282c0]
                                                            1  _NSXPCSerializationStringForObject + 51 (Foundation + 139625) [0x7ff80468e169]
                                                              1  _getStringAtMarker + 440 (Foundation + 1110832) [0x7ff80477b330]
                                                                1  CFStringCreateWithCharacters + 41 (CoreFoundation + 62887) [0x7ff8036c25a7]
                                                                  1  __CFStringCreateImmutableFunnel3 + 860 (CoreFoundation + 11217) [0x7ff8036b5bd1]
                                                                    1  <on behalf of UNKNOWN [40261], User mode, Effective Thread QoS Default, Requested Thread QoS Default>
  1  start_wqthread + 0 (libsystem_pthread.dylib + 7048) [0x7ff80364eb88]
    1  <Effective Thread QoS User Interactive, Requested Thread QoS User Interactive>

  Binary Images:
           0x109a4f000 -        0x109a72fff  com.latenitefilms.GyroflowToolbox.Renderer 1.1.4 (18) <8CE8B05F-A555-3085-9893-49775742B5F6>  /Applications/Gyroflow Toolbox.app/Contents/PlugIns/Gyroflow Toolbox Renderer.pluginkit/Contents/MacOS/Gyroflow Toolbox Renderer
[... cut ...]
        0x7ff8036b3000 -     0x7ff803b4dfec  com.apple.CoreFoundation 6.9 (2303)                   <D9705B29-47AF-343B-BA26-5CDA3166731E>  /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
        0x7ff80466c000 -     0x7ff8052a3ff3  com.apple.Foundation 6.9 (2303)                       <6589CC92-B39E-392C-808B-B20538F9312F>  /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation 

</details>

PS. i didn't find any gyroflow log files in Group Containers

latenitefilms, 2024-05-01:
You can access the log files via this button in the Final Cut Pro Inspector:

[image]

yarovikov, 2024-05-01:
thanks 😀

[GyroflowCore.log](https://github.com/gyroflow/gyroflow/files/15176042/GyroflowCore.log)
[FxPlug.log](https://github.com/gyroflow/gyroflow/files/15176043/FxPlug.log)

latenitefilms, 2024-06-25:
@AdrianEddy - Does this mean anything to you?

```
thread '<unnamed>' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_shader_module
    
Shader '' parsing error: redefinition of `uv`
    ┌─ wgsl:119:36
    │
119 │         fn digital_undistort_point(uv: vec2<f32>) -> vec2<f32> {
    │                                    ^^ previous definition of `uv`
120 │             let out_c2 = vec2<f32>(f32(params.output_width), f32(params.output_height));
121 │             var uv = uv;
    │                 ^^ redefinition of `uv`

    redefinition of `uv`

', /Users/chrishocking/.cargo/git/checkouts/wgpu-53e70f8674b08dd4/359e392/wgpu/src/backend/direct.rs:3095:5
thread '<unnamed>' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_shader_module
    
Shader '' parsing error: redefinition of `uv`
    ┌─ wgsl:119:36
    │
119 │         fn digital_undistort_point(uv: vec2<f32>) -> vec2<f32> {
    │                                    ^^ previous definition of `uv`
120 │             let out_c2 = vec2<f32>(f32(params.output_width), f32(params.output_height));
121 │             var uv = uv;
    │                 ^^ redefinition of `uv`

    redefinition of `uv`

', /Users/chrishocking/.cargo/git/checkouts/wgpu-53e70f8674b08dd4/359e392/wgpu/src/backend/direct.rs:3095:5
thread '<unnamed>' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_shader_module
    
Shader '' parsing error: redefinition of `uv`
    ┌─ wgsl:119:36
    │
119 │         fn digital_undistort_point(uv: vec2<f32>) -> vec2<f32> {
    │                                    ^^ previous definition of `uv`
120 │             let out_c2 = vec2<f32>(f32(params.output_width), f32(params.output_height));
121 │             var uv = uv;
    │                 ^^ redefinition of `uv`

    redefinition of `uv`

', /Users/chrishocking/.cargo/git/checkouts/wgpu-53e70f8674b08dd4/359e392/wgpu/src/backend/direct.rs:3095:5
thread '<unnamed>' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_shader_module
    
Shader '' parsing error: redefinition of `uv`
    ┌─ wgsl:119:36
    │
119 │         fn digital_undistort_point(uv: vec2<f32>) -> vec2<f32> {
    │                                    ^^ previous definition of `uv`
120 │             let out_c2 = vec2<f32>(f32(params.output_width), f32(params.output_height));
121 │             var uv = uv;
    │                 ^^ redefinition of `uv`

    redefinition of `uv`

', /Users/chrishocking/.cargo/git/checkouts/wgpu-53e70f8674b08dd4/359e392/wgpu/src/backend/direct.rs:3095:5
thread '<unnamed>' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_shader_module
    
Shader '' parsing error: redefinition of `uv`
    ┌─ wgsl:119:36
    │
119 │         fn digital_undistort_point(uv: vec2<f32>) -> vec2<f32> {
    │                                    ^^ previous definition of `uv`
120 │             let out_c2 = vec2<f32>(f32(params.output_width), f32(params.output_height));
121 │             var uv = uv;
    │                 ^^ redefinition of `uv`

    redefinition of `uv`

', /Users/chrishocking/.cargo/git/checkouts/wgpu-53e70f8674b08dd4/359e392/wgpu/src/backend/direct.rs:3095:5
thread '<unnamed>' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_shader_module
    
Shader '' parsing error: redefinition of `uv`
    ┌─ wgsl:119:36
    │
119 │         fn digital_undistort_point(uv: vec2<f32>) -> vec2<f32> {
    │                                    ^^ previous definition of `uv`
120 │             let out_c2 = vec2<f32>(f32(params.output_width), f32(params.output_height));
121 │             var uv = uv;
    │                 ^^ redefinition of `uv`

    redefinition of `uv`

', /Users/chrishocking/.cargo/git/checkouts/wgpu-53e70f8674b08dd4/359e392/wgpu/src/backend/direct.rs:3095:5
thread '<unnamed>' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_shader_module
    
Shader '' parsing error: redefinition of `uv`
    ┌─ wgsl:119:36
    │
119 │         fn digital_undistort_point(uv: vec2<f32>) -> vec2<f32> {
    │                                    ^^ previous definition of `uv`
120 │             let out_c2 = vec2<f32>(f32(params.output_width), f32(params.output_height));
121 │             var uv = uv;
    │                 ^^ redefinition of `uv`

    redefinition of `uv`

', /Users/chrishocking/.cargo/git/checkouts/wgpu-53e70f8674b08dd4/359e392/wgpu/src/backend/direct.rs:3095:5
thread '<unnamed>' panicked at 'wgpu error: Validation Error

Caused by:
    In Device::create_shader_module
    
Shader '' parsing error: redefinition of `uv`
    ┌─ wgsl:119:36
    │
119 │         fn digital_undistort_point(uv: vec2<f32>) -> vec2<f32> {
    │                                    ^^ previous definition of `uv`
120 │             let out_c2 = vec2<f32>(f32(params.output_width), f32(params.output_height));
121 │             var uv = uv;
    │                 ^^ redefinition of `uv`

    redefinition of `uv`

', /Users/chrishocking/.cargo/git/checkouts/wgpu-53e70f8674b08dd4/359e392/wgpu/src/backend/direct.rs:3095:5
```

AdrianEddy, 2024-06-25:
> @AdrianEddy - Does this mean anything to you?

Yes, but that is already fixed in master

latenitefilms, 2024-06-25:
@yarovikov - Apologies for the delay! This will hopefully be fixed in the Gyroflow Toolbox v1.1.5 update, which should hit the Mac App Store soon.

AdrianEddy, 2024-09-26:
since there was no further reports after v1.1.5 release, I consider this fixed

closed 2024-09-26 by AdrianEddy