gyroflow/gyroflow #815: Does an extension tube affect lens profile?

xphoniex, 2024-04-17:
I'm trying to fix rolling shutter effect of some macro footage, and not only I'm not getting improvements where there's jello effect, it's also distorting the image in other areas.

I've seen some impressive results on YouTube for wide shots, so need to know what am I doing wrong here, is it:
- macro/closeup can't be fixed, contrary to wide?
- wrong lens profile due to having an extension tube?
- because of 180° shutter?

Here's an example:

[image]

You can see the straight line bulging in the gyroflow output.

AdrianEddy, 2024-04-17:
Yes, the extension tube definitely affects the lens profile. You should calibrate your own profile with the exact same setup

xphoniex, 2024-04-17:
Is there any guide how to do that written somewhere?

Also, any comments re jello effect? I'm not familiar with the physics, should I be expecting as good a fix for macro as the wide shots?

AdrianEddy, 2024-04-17:
https://docs.gyroflow.xyz/app/getting-started/lens-calibration

> Also, any comments re jello effect? I'm not familiar with the physics, should I be expecting as good a fix for macro as the wide shots?
In ideal conditions yes, but hard to tell without seeing the exact sample. Definitely first step is the accurate lens profile and rolling shutter value. What camera are you using?

xphoniex, 2024-04-17:
> https://docs.gyroflow.xyz/app/getting-started/lens-calibration

Hm.. it'd be very hard, if not impossible, to get the whole chessboard pattern in one shot as the macro FOV is small, and so is the focus plane. What's your suggestion?

> In ideal conditions yes, but hard to tell without seeing the exact sample. Definitely first step is the accurate lens profile and rolling shutter value. What camera are you using?

I didn't know how to cut a small portion of video while retaining gyro data. What settings should I use? (in dr or ffmpeg)

By rolling shutter value, you mean the ms value? I tried a few variations which didn't make much of a difference.

Using ZV-E10 with kit lens.

AdrianEddy, 2024-04-17:
> Hm.. it'd be very hard, if not impossible, to get the whole chessboard pattern in one shot as the macro FOV is small, and so is the focus plane. What's your suggestion?

Just scale down this pattern and print it, then film the small one on a paper. That should work. Make it so the entire pattern is visible in the lens, but the physical size doesn't matter, so it can be really small

> I didn't know how to cut a small portion of video while retaining gyro data. What settings should I use? (in dr or ffmpeg)

it's not possible, you need to use the entire file

> By rolling shutter value, you mean the ms value?

Yes, but with ZV-E10 it's already correct from the video metadata, don't change it

closed 2024-04-18 by xphoniex