gyroflow/gyroflow #816: Video blank if gyro data is imported but not if just the video is imported and not the gyro data

CrashDBN, 2024-04-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4(gh1826)

### What operating system are you using?

Windows 11

### What GPU are you using?

AMD Radeon Vega 8

### What happened?

If I just import the video file from the Hawkeye Split 4k V4 I can see the video etc and its not blank if I then go and import the gyro data the video goes black
[image]
[image]

AdrianEddy, 2024-04-17:
go to Advanced on bottom right and "Reset all settings to defaults"

CrashDBN, 2024-04-17:
Hi there.Thried that removed the app and reinstalled same thing Get Outlook for Android

AdrianEddy, 2024-04-17:
show me a screenshot of the "Stabilization" section opened

CrashDBN, 2024-04-17:
[image]

CrashDBN, 2024-04-17:
Hi there

I came across this issue it is sorta related to my issue and that is why I am using the dev version of gyroflow but it said it was resolved. So not sure if its got anything to do with it.

https://github.com/gyroflow/gyroflow/issues/559

AdrianEddy, 2024-04-17:
1. Install latest GPU driver (I think it may be this one https://dl.dell.com/FOLDER07839672M/1/AMD-Radeon-Graphics-Driver_251TV_WIN64_30.0.13020.4_A01.EXE )
2. If that doesn't help, send me gyroflow.log file (it's next to Gyroflow.exe)

CrashDBN, 2024-04-17:
[image]

closed 2024-04-17 by CrashDBN