gyroflow/gyroflow #817: Lens profile loading error with Insta360 Ace Pro footage

webboy634, 2024-04-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.4

### What operating system are you using?

Windows 11 Home OS Build 22631.3447

### What GPU are you using?

NVIDIA GeForce RTX 4070 Ti

### What happened?

Cannot get Insta360 Ace Pro freeform video to load in Gyroflow. If I just try to open a file, I get a VERY quick message across the top in a yellow bar about Lens Profile not loading and then the program crashes.

There are no Insta360 Ace Pro lens profile installed with the application so that's not a possibility.
[image]

AdrianEddy, 2024-04-19:
it's already fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2024-04-19 by AdrianEddy

huntkao, 2024-07-17:
I can successfully build and run the master code. How should I modify the code to make Insta360 Ace Pro be supported as 1.6.0?
Million thanks!

AdrianEddy, 2024-07-17:
It's already supported, what issue do you have?

webboy634, 2024-07-17:
Was the older version. Everything's good now..ticket is closed.