gyroflow/gyroflow #818: Setting QT version during build (linux)

siefkenj, 2024-04-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

git head

### What operating system are you using?

gentoo x86_64

### What happened?

I am attempting to build on a Gentoo Linux system. I believe I have all deps installed. My system is has Qt 6.7.0 installed, but `cargo run --release` fails when compiling `qml-video-rs` as it is trying to include Qt 6.5.3.

```
error occurred: ToolExecError: Command "c++" "-O3"
 "-ffunction-sections" "-fdata-sections" "-fPIC"
 "-m64" "-I" "/xxx/.cargo/git/checkouts/qml-video-rs-a29d2c8701d704e5/520acc0" 
"-I" "/usr/include/qt6/QtCore" "-I" "/usr/include/qt6/QtGui" "-I"
 "/usr/include/qt6/QtQuick" "-I" "/usr/include/qt6/QtQml" 
"-I" "/usr/include/qt6/QtCore/6.5.3"
...
```
I have tried setting the `DEP_QT_VERSION=6.7.0` env variable, but that doesn't seem to change the Qt include flags. Is there a different way to set the version? (If I copy and paste the failed command and substitute 6.5.3 for 6.7.0, that command succeeds. But of course that is out of the build processes, so doesn't help me that much...)

AdrianEddy, 2024-04-20:
Looks like your c++/clang is too old? I don't think it has anything to do with Qt version. Also, it says 6.5.3 in the path?

siefkenj, 2024-04-20:
`just run` does the same thing as `cargo run` on linux, at least if I understand the `just` script properly. `c++` is currently set to `gcc-12.3`. If I manually switch out all the references to `6.5.3` for `6.7.0`, the failing command builds fine, so I don't think its an outdated toolchain issue.

I have no v6.5.3 installed on my system, so I don't know where it's pulling that number from...

AdrianEddy, 2024-04-20:
`just run` is not the same as `cargo run`, the most important thing that it does is to set environment variables to point to correct dependencies versions (Qt, ffmpeg, OpenCV, mdk-sdk)

6.5.3 is nowhere in the sources, so it must be the `qmake` in your PATH that causes it. Make sure you run `just run` or set `PATH` proeperly

siefkenj, 2024-04-21:
I have also been running `just run` with the same effect. Can you show me where `just` configures the env vars? I only see
```
[no-cd]
run *param:
    cargo run --release -- {{param}}
```
in `_scripts/linux.just`. Maybe if I can get at the configured variables, I can figure out who is missetting them...I also don't have a `qmake` on my system...

AdrianEddy, 2024-04-21:
`common.just` is included at the beginning which sets all the variables first, it's a very important step

siefkenj, 2024-04-21:
I am not on a debian-based system, so `just install-deps` won't work for me (no `apt-get`). I have all the required libraries already installed on my system, though I am now running into trouble with OpenCV...

Executing
```
OPENCV_CMAKE_NAME=OpenCV OPENCV_CMAKE_BIN=`which cmake` LIBCLANG_PATH=/usr/lib/llvm/17/lib64/ just run
```
seems to have no effect on the builds ability to find OpenCV, though if I run
```
OPENCV_CMAKE_NAME=OpenCV OPENCV_CMAKE_BIN=`which cmake` LIBCLANG_PATH=/usr/lib/llvm/17/lib64/ cargo run --release
```
it finds OpenCV just fine, though I get a `linking with cc failed: exit status 1` on a hugely long call to the linker.

siefkenj, 2024-04-21:
I've made a little bit of progress. I installed `OpenCV 4.8.0` via `vcpkg` into `gyroflow/ext` and now `LIBCLANG_PATH=/usr/lib/llvm/17/lib64/ just run` terminates with
```
/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lpng
```
which is strange because directly running
```
/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/ld -lpng --verbose|grep libpng
```
shows `attempt to open /lib64/libpng.so succeeded`, which means that that version of `ld` is definitely capable of finding `libpng`.

AdrianEddy, 2024-04-21:
it would be the easiest to just replace `apt` calls to your system package manager and leave everything else as is and use the just script. I takes care of all the proper environment variables, the Qt version, ffmpeg version and OpenCV version.

The `apt` calls are very minimal, just install these packages your way and leave everything else to the `just` script

siefkenj, 2024-04-21:
I'm a bit confused...must of what is done in `just install-deps` seems to be Debian specific, and `just install-deps` should not leave any residual env vars set. So, given that I have all the packages installed that `just install-deps` would install, running `just run` should set up the env vars that you're talking about, right?

If you had the time to do a screen share, I could show you, and maybe you'd understand the issue.

AdrianEddy, 2024-04-21:
> I'm a bit confused...must of what is done in just install-deps seems to be Debian specific,

These are the only debian-specific lines:
```
    sudo apt-get install -y p7zip-full python3-pip clang libclang-dev bison pkg-config gperf curl unzip zip git
    sudo apt-get install -y libc++-dev libva-dev libvdpau-dev libvdpau1 mesa-va-drivers ocl-icd-opencl-dev opencl-headers
    sudo apt-get install -y libpulse-dev libasound-dev libxkbcommon-dev
    sudo apt-get install -y libfontconfig1 libfreetype6
```

Everything else is universal

> and just install-deps should not leave any residual env vars set.

It doesn't, it installs all deps to the `ext` folder

>  So, given that I have all the packages installed that just install-deps would install, running just run should set up the env vars that you're talking about, right?

Yes, but they have to be in the `ext` folder, not on your system

siefkenj, 2024-04-22:
I have followed your suggestion in a fresh clone of the repo. I commented out all the `sudo apt-get` lines in `linux.just` and ran `just install-deps` followed by `just run`. Since the build system could not find `libclang.so`, I had to set `LIBCLANG_PATH`, so the command I actually ran was:

```
LIBCLANG_PATH="/usr/lib/llvm/16/lib64" just run
```
I get an error when building `gyroflow`:
```
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lpng
```
I have tried both GCC v12 and v13, as well as linking to `libclang.so` v16 and v17.

Any ideas why it might be failing to resolve libpng (which is installed on my system, as I showed before...)

AdrianEddy, 2024-04-22:
install [libpng](https://packages.gentoo.org/packages/media-libs/libpng) for your system

siefkenj, 2024-04-22:
> install [libpng](https://packages.gentoo.org/packages/media-libs/libpng) for your system

Like I said, `libpng` is installed and recognized by `ld`. 

```
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld -lpng --verbose|grep libpng
attempt to open /lib64/libpng.so succeeded
/lib64/libpng.so
```

AdrianEddy, 2024-04-22:
but you have the shared binary installed (.so). You need the static library for libpng - the `libpng.a` file. On debian, these packages have the  `-dev` suffix

siefkenj, 2024-04-22:
Thank you! Installing with `static-libs` got past the `-lpng` error :-).

There's a new one now, though...
```
/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/ld: /home/albert/downloads/gyroflow2/target/release/build/qml-video-rs-600d545abebb22ba/out/mdk-sdk/lib/amd64//libmdk.so: undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::append(char const*, unsigned long)'
```
I checked and `libmdk.so.0` is in that folder.

AdrianEddy, 2024-04-22:
this error indicates some mismatch in c++ standard library, possibly because of different clang versions used? Try `cargo clean` and `just run` again

siefkenj, 2024-04-23:
`cargo clean` didn't help, but it turned out I needed to install Gentoo's `sys-libs/libcxx`. Now there's progress! I get the message
```
 warning: libwayland-client.so.0, needed by /home/albert/downloads/gyroflow2/target/release/build/qml-video-rs-600d545abebb22ba/out/mdk-sdk/lib/amd64//libmdk.so, not found 
```
The entire reason I am trying to build this package myself is because the AppImage fails due to me no having Wayland. Is there a way to build without wayland?

AdrianEddy, 2024-04-23:
This looks like a warning, are you sure it's that message that's stopping your build? Wayland shouldn't required. Also, did you try the prebuilt .tar.gz package from the website?

siefkenj, 2024-04-23:
Yes, I tried the prebuilt package. I ended up installing `xwayland` and after that, the gyroflow starts!

siefkenj, 2024-04-23:
I'm going to try to produce a Gentoo `ebuild` for gyrflow. Hopefully this thread can be useful to some others.

AdrianEddy, 2024-04-23:
it'd be better to add an `if` to the debian specific lines and add gentoo-specific lines there

closed 2024-05-10 by AdrianEddy