gyroflow/gyroflow #819: (Android build) Preview on S22 Ultra Android 14 does not play with videos 5min+

Strubu, 2024-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

Android 14

### What GPU are you using?

S22 Ultra

### What happened?

As in the title: Previews on S22 Ultra Android 14 do not play with videos over 5min. I tested short videos and they work flawless. Also tried with a lent Pixel 8, and the same 5min+ videos preview without issue (even though rendering is slower on the Pixel 8 :D).
Any ideas how to fix it?

AdrianEddy, 2024-04-21:
are you sure the 5 min is the main factor? what camera is the video file from? does it have gyro data? what resolution, codec?

AdrianEddy, 2024-04-24:
Ok go to Advanced and try different "Preview pipeline". Also does anything change when you disable stabilization?

Strubu, 2024-04-24:
> Ok go to Advanced and try different "Preview pipeline". Also does anything change when you disable stabilization? [...]

Tried all combinations of preview settings. Preview also stops even if stabilization is switched off. Video plays normal with the build in video player, so I don't think, it's a codec problem either.

AdrianEddy, 2024-04-24:
Try this apk https://github.com/gyroflow/gyroflow/actions/runs/8444901910/artifacts/1361449368

Strubu, 2024-04-24:
> Try this apk https://github.com/gyroflow/gyroflow/actions/runs/8444901910/artifacts/1361449368

Tried the apk, but same behaviour. 2.7k videos run in preview with no issues, 4k videos run for 2-3 frames, then lock up. 
Pixel 8 plays all without issues. It doesn't matter if I choose the "[wgpu] Adreno(tm) 730 (Vulkan)" or "CPU" option for the preview pipeline video processing, else I would have blamed it on the S22 GPU architecture. I'm going to make some anonymized videos tomorrow morning and post them for testing. Thanks for the quick response.

Strubu, 2024-04-25:
> Try this apk https://github.com/gyroflow/gyroflow/actions/runs/8444901910/artifacts/1361449368

Here are 3 videos (4k 4:3 not working, 4k 16:9 working, 2.7k 4:3 working) and some screen recordings. 
https://1drv.ms/f/s!AjwietZirFfNkOwZ35wYcr89FM9aVg?e=VMDadN

AdrianEddy, 2024-04-26:
How about this apk? https://github.com/gyroflow/gyroflow/actions/runs/8849189979/artifacts/1451192994

Strubu, 2024-04-26:
> How about this apk? https://github.com/gyroflow/gyroflow/actions/runs/8849189979/artifacts/1451192994

Still same behavior.
Btw: I just realized, that scrubbing though the whole video is possible, before pressing play. After pressing play, the video locks up again after a few frames. I added a new screen recording to the OneDrive, using the new apk.

AdrianEddy, 2024-05-23:
Can you try latest dev build? https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-android.zip

AdrianEddy, 2024-05-29:
Any update? did you try the dev build?

Strubu, 2024-06-10:
> Any update? did you try the dev build?

Sorry, was away for a while. While on vacation, I installed Gyroflow on my old Samsung S10 and ran the 4k Avata2 files without problems from there (export took forever on that phone, though :D ). 

On the S22 Ultra, I tried the latest dev-build you uploaded. It shows exactly the same behavior as before (preview crashes on pressing play; scrubbing through the whole video before that is possible) 

Might be a niche problem with the Exynos 2200 chip (I have the Model SM-S908U1)?

RCFPV, 2025-02-13:
Hi. Was this problem with gyroflow crashing when previewing 4k 4:3 ever resolved. I am trying to do it on my S23 and Samsung Tab S8 without the luck :(

AdrianEddy, 2025-02-13:
I don't know, did you try the dev build apk? https://gyroflow.xyz/devbuild/

medk42, 2025-03-08:
Same issue here in the tab s9. yuv420 4k 16:9 10bit from o4 pro works. But yuv420 4k 8:7 10bit from gp12 doesn't. It can render, and i can scroll through the preview,  but I can't play the preview,  everything gets stuck if i try. The nightly build server isn't working currently, so we'll see later, maybe.

medk42, 2025-03-08:
The nightly build server works now, I downloaded the nightly build - same issue. The 16:9 dji video is fine, but the 8:7 gopro log high bitrate 10bit video can be scrolled through, but not played. When I play, it plays about 1sec and then gets stuck and I have to restart the app (scrolling preview no longer works either, but I can still export) - so it does behave differently, but still doesn't play.