gyroflow/gyroflow #821: Inquiry Regarding Implementation Details of lock_horizon_angle Function in horizon.rs

tigercc8, 2024-04-22:
Dear everyone

I hope this email finds you well. I am a student currently working on my graduation project. As part of my project, I have been exploring the Gyroflow codebase, particularly the horizon.rs file, to gain a deeper understanding of its functionality.

While reviewing the code, I encountered the lock_horizon_angle function, and I must admit that I'm having some difficulty grasping its implementation logic, especially regarding the angle calculation method and the initial_quat parameter. I was wondering if you could kindly provide some clarification or insight into how this function is structured and how it achieves its intended purpose.

Specifically, I would appreciate it if you could elaborate on:

1. The approach used to calculate the angle in the lock_horizon_angle function. From：https://github.com/gyroflow/gyroflow/commit/2fc7cf7f15d103f9c6dbab18849937049114df13
2. The significance and role of the initial_quat parameter within the function.
3. Any additional details or considerations that would aid in understanding the implementation logic of this function.
Your expertise and guidance would be immensely valuable to me as I strive to comprehend the intricacies of Gyroflow for my graduation project. Any insights or explanations you could offer would greatly contribute to my learning process.

Thank you very much for considering my inquiry. I eagerly await your response and appreciate your time and assistance in advance.

Warm regards,

AdrianEddy, 2024-04-22:
@mycosd @ElvinC @VladimirP1 any of you knows more about the math here than me

mycosd, 2024-04-22:
´lock_horizon_angle´ takes the quaternion with of a camera rotation and then gets the yaw & pitch angles of it. It does it by creating a vector in the 3D space using the quaternion and an axis and then measuring its angles in relation to the axis's. Then it creates a new quaternion with the extracted yaw and pitch angles and also the new requested roll angle (completely ignores the original roll). 

The `initial_quat` is the base rotation that the quaternions that gyroflow expects are referenced to. So essentially if a quaternion is equal to initial_quat, the camera is not rotated at all. At least that's what I understood when refactoring it from the original code.

tigercc8, 2024-04-22:
> `lock_horizon_angle` takes the quaternion of a camera rotation and then gets the yaw & pitch angles of it. It does it by creating a vector in the 3D space using the quaternion and an axis and then measuring its angles in relation to the axis's. Then it creates a new quaternion with the extracted yaw and pitch angles and also the new requested roll angle (completely ignores the original roll).
> 
> The `initial_quat` is the base rotation that the quaternions that gyroflow expects are referenced to. So essentially if a quaternion is equal to initial_quat, the camera is not rotated at all. At least that's what I understood when refactoring it from the original code.

Thank you very much for your reply. I know what the process looks like. But what puzzles me is why multiply the input `quaternion` by the `z-axis` unit vector? I'm not sure what coordinate system conversion is like in the `lock_horizon_angle function`. I'm a little confused between the `xyz axis` and `yaw, roll and pitch`. Could you explain it in more detail? Thank you very much!

mycosd, 2024-04-22:
> Thank you very much for your reply. I know what the process looks like. But what puzzles me is why multiply the input `quaternion` by the `z-axis` unit vector?

Essentially think of the z-axis as the axis going through the center of your camera image to the horizon, when the camera is not rotated at all. Now you have a rotation as a quaternion, so you rotate the z-axis with that quaternion, so the new vector points in the direction the camera is actually pointing to. If you want an angle of rotation of your camera, you calculate it from the z-axis (zero rotation) and the new vector (full rotation). You can also use the other axis's as they are in zero rotation and have a known angle in relation to the z-axis.

tigercc8, 2024-04-22:
> If you're asking about why this particular orientation was chosen, then I think no one has an answer for that. It was just a historic choice and it went through a few iterations when developing, starting with being based on GoPro convention, then as we started to add more gyro sources (I think the next was Betaflight blackbox), and then horizon lock was added a bit later on top of that without changing the original code, so it's all a bit of a mess. It was never designed from the ground up with a horizon lock in mind

Thank you for your reply!  Where can I find a reference to GoPro convention? Or is this just gyroflow's historic choice? I've never been sure what the coordinate system inside gyroflow is, just going along with my guess, need to be sure.

tigercc8, 2024-04-22:
> Essentially think of the z-axis as the axis going through the center of your camera image to the horizon, when the camera is not rotated at all. Now you have a rotation as a quaternion, so you [...]

> > Thank you very much for your reply. I know what the process looks like. But what puzzles me is why multiply the input `quaternion` by the `z-axis` unit vector?
> 
> Essentially think of the z-axis as the axis going through the center of your camera image to the horizon, when the camera is not rotated at all. Now you have a rotation as a quaternion, so you rotate the z-axis with that quaternion, so the new vector points in the direction the camera is actually pointing to. If you want an angle of rotation of your camera, you calculate it from the z-axis (zero rotation) and the new vector (full rotation). You can also use the other axis's as they are in zero rotation and have a known angle in relation to the z-axis.
> 
> Edit: Also you have to take the `initial_quat` rotation into a account, so if you rotate the z-axis with the `initial_quat` which represents a zero rotation, the axis's get a bit mangled.

After listening to your explanation, I seem to understand the reason for doing so. I mistakenly think that the `X-axis` is the axis pointing to the horizon. But I still have a question: `initial_quat` is intended to be needed to convert the rotation to gyroflow, and if the horizon lock function is not selected, there will be no conversion process, followed directly by smoothing the original quaternion. Does the rotation vector of the `quaternion` multiplied by the `z axis` have the effect of rotating the coordinate system such that `initial_quat` is required to restore it?

mycosd, 2024-04-22:
Gyroflow for most of its functionality doesn't care about the base rotation, as everything is relative. For Horizon lock, you have real world reference (horizon), so that's why you'll have to disect the quaternions.
The `quaternion * z-Axis` creates a vector with the length of 1 pointing somewhere, and you can use it get angles. As `quaternion` in this case also contains the `initial_quat` rotation you can think of the whole angle measuring like this:
1. create the test-vector `quaternion * z-Axis`
2. rotate your xyz coordinate system by `initial_quat`
3. measure the angles of your test-vector relative to the rotated coordinate system
Step 2 is a bit of hidden in the code, that's why it appears we are using the wrong vector components or use negative values for z out of nowhere.

mycosd, 2024-04-22:
For step 2, just imagine what would happen if you pass eg. the z-Axis vector through the `initial_quat` rotation. First it's rotated 90° around the y-Axis, so it would virtually become the x-Axis, then it's rotated 90° around the z-Axis and becomes virtually an inverted y-Axis. Do the same for the other axis's, and then when extracting the angles, interprete the xyz-components of the vector accordingly. So if you need eg. the z-component, you'll have to read -y.

tigercc8, 2024-04-22:
> For step 2, just imagine what would happen if you pass eg. the z-Axis vector through the `initial_quat` rotation. First it's rotated 90° around the y-Axis, so it would virtually become the x-Axis, [...]

Thank you！I get it now！

closed 2024-04-24 by tigercc8

ysyscool, 2024-07-15:
> `lock_horizon_angle` takes the quaternion of a camera rotation and then gets the yaw & pitch angles of it. It does it by creating a vector in the 3D space using the quaternion and an axis and then [...]

Hi,

May I know how to get the new requested roll angle? I didn't find the relevant code about it. Is it loaded from the data？Or calculated by the algorithm?