gyroflow/gyroflow #823: mkv file format not supported

siefkenj, 2024-04-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

git head

### What operating system are you using?

gentoo x86_64

### What happened?

Loading an mkv video, the video appears partially cropped in the player and "Auto sync" fails to work. 

[image]

The same video in an mp4 container works perfectly. Attached is an mp4 and mkv sample (created with AviDemux)

[sample_960x540.zip](https://github.com/gyroflow/gyroflow/files/15098522/sample_960x540.zip)

siefkenj, 2024-04-24:
I have video that has no gyro data that I have trimmed and saved as in the mkv format. I could convert everything to mp4 and then apply gyroflow, but it seem like mkv support might not be hard to add.

AdrianEddy, 2024-04-24:
You need to load a lens profile to be able to click autosync for optical-only analysis

AdrianEddy, 2024-04-24:
@wang-bin looks like this sample mkv file doesn't report duration in mdk, any idea why?

wang-bin, 2024-04-26:
> @wang-bin looks like this sample mkv file doesn't report duration in mdk, any idea why?

which duration do you read? duration from container is correct

```
Format: matroska,webm, range: 0 +13346ms, bitrate: 303214, size: 0
Metadata:
  ENCODER: Lavf58.76.100
Streams: 1
 Video:
  stream#0, range: 0 +0ms, frames: 0
  codec: h264 tag: 0 profile: 100 level: 31, yuv420p, bpc:8, bpp:12(8,8,8), channels:(1,1,1), map: 0 1 2 0, bitrate: 0, 960x540, fps: 29.97, bframes: 0
  extra data(55): 01 64 00 1F FF E1 00 24 67 64 00 1F AC 2B 20 1E 02 2F DE 02 20 00 00 7D 20 00 1D 4C 1C 00 00 03 01 55 CC 00 00 0D 59 F8 5D E5 C1 40 01 00 04 68 EB 8F 2C FD F8 F8 00 
  Metadata:
   DURATION: 00:00:13.346000000
```

AdrianEddy, 2024-04-26:
@wang-bin I'm using `player->mediaInfo().video[0].duration`

wang-bin, 2024-04-27:
> @wang-bin I'm using `player->mediaInfo().video[0].duration`

the latest build will read duration from metadata

AdrianEddy, 2024-04-28:
thanks, I can confirm it works fine in latest dev build: https://gyroflow.xyz/devbuild/

closed 2024-04-28 by AdrianEddy