gyroflow/gyroflow #825: Full Sony metadata support

AdrianEddy, 2024-04-26:
In order: 
- [ ] #44
- [ ] #727 
- [ ] OIS data (tag [0xe416](https://github.com/AdrianEddy/telemetry-parser/blob/master/src/sony/rtmd_tags.rs#L450)) I think this means adding this data to cx and cy of the camera intrinsic matrix, but I could be wrong
- [ ] EIS data ([already figured out](https://github.com/gyroflow/gyroflow/files/15135551/mesh_correction_reverse.zip), just needs to be ported to the gpu kernel)

AdrianEddy, 2024-04-26:
/bounty $10000

joyboy50, 2024-04-27:
/attempt #825

digital-phoenix, 2024-04-28:
/attempt #825

feliciien, 2024-04-29:
/attempt #82

Ahmadkhan02, 2024-05-01:
/attempt #825

Rutik7066, 2024-05-08:
/attempt

Subh231004, 2024-06-09:
Attempt #825

the-johnwick, 2024-06-25:
/attempt

AdrianEddy, 2024-06-26:
This is now mostly done. Majority of the work was done by @VladimirP1 and me (EIS), with reverse-engineering help from @digital-phoenix.

If you're curious what this task required, it's essentially [sony.rs](https://github.com/gyroflow/gyroflow/blob/master/src/core/gyro_source/sony.rs) + changes in [frame_transform.rs](https://github.com/gyroflow/gyroflow/blob/master/src/core/stabilization/frame_transform.rs#L245-L263) + changes in kernels eg. in opencl_undistort.cl [ibis/ois](https://github.com/gyroflow/gyroflow/blob/master/src/core/gpu/opencl_undistort.cl#L302-L332) and [eis](https://github.com/gyroflow/gyroflow/blob/master/src/core/gpu/opencl_undistort.cl#L170-L228)

The amount of work required to get this done was insane so please pay huge respect to @VladimirP1, he figured out all the needed math for this, and @digital-phoenix did a lot of reverse-enginnering with Frida scripts required to dump data from Catalyst and compare logic and functions. 

There are still some bugs in some files, but majority of the videos already work properly. For the remaining issues see #849 

/tip $1000 @digital-phoenix

closed 2024-06-26 by AdrianEddy

rgaufman, 2024-06-26:
Are there any instructions how to use this? Or is it just automatic in the latest version from git?

AdrianEddy, 2024-06-26:
automatic https://gyroflow.xyz/devbuild/

danielkelNL, 2024-06-28:
Thank you for all the time and effort! Does it also work in combination with Active Stabilization mode. The Sony A7IV does have this mode.

AdrianEddy, 2024-06-28:
Yes

Hevinxue, 2024-07-09:
> automatic https://gyroflow.xyz/devbuild/

Tried the beta and:
"0xc00000142" error on 2 of my 3 windows platforms, win11(23H2) and win10(22H2).

AdrianEddy, 2024-07-09:
@Hevinxue Install VC Redist https://aka.ms/vs/17/release/vc_redist.x64.exe

cesccc, 2024-07-14:
Is the bounty still on? @AdrianEddy

cyfdecyf, 2024-08-14:
> Yes, try it

I'm really excited when I see this message. I've tried the dev build with some videos shoot on Sony A7 IV with PZ 16-35 lens at various focal lengths (with IBIS turned on), all works very well automatically like magic. The stablization even works for videos that has zooming.

Thanks for the great work!

rgaufman, 2024-09-25:
Total game changer this and so much faster and better than Sony's Catalyst!

yamancan, 2024-10-12:
/attempt 825