gyroflow/gyroflow #826: ffmpeg error for the export

Surfoo, 2024-04-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

Manjaro

### What GPU are you using?

MSI RX 550 AERO ITX 2G

### What happened?

Hello,

For some videos, I have this error message. The GPU is not used, and I use H264 for the export.
What's going wrong?

Thanks! 

[image]

### Relevant log output

```shell
15:55:07 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: None
15:55:07 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: None
[mp4 @ 0x7d7198d9d3c0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 1345475 >= 1345475
15:55:07 [ERROR] [mp4 @ 0x7d7198d9d3c0] Application provided invalid, non monotonically increasing dts to muxer in stream 0: 1345475 >= 1345475
15:55:07 [DEBUG] (2) gyroflow_core::filesystem: FfmpegPathWrapper::drop file:///mnt/westernd/ski2024/medias/20240407_140957956.TS_stabilized_1.mp4
[libx264 @ 0x7d7198963080] frame I:17    Avg QP: 6.90  size:232134
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] frame I:17    Avg QP: 6.90  size:232134
[libx264 @ 0x7d7198963080] frame P:157   Avg QP: 9.58  size:115324
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] frame P:157   Avg QP: 9.58  size:115324
[libx264 @ 0x7d7198963080] frame B:276   Avg QP:12.02  size: 55124
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] frame B:276   Avg QP:12.02  size: 55124
[libx264 @ 0x7d7198963080] consecutive B-frames: 15.5%  5.8%  5.3% 73.5%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] consecutive B-frames: 15.5%  5.8%  5.3% 73.5%
[libx264 @ 0x7d7198963080] mb I  I16..4: 15.0% 58.9% 26.0%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] mb I  I16..4: 15.0% 58.9% 26.0%
[libx264 @ 0x7d7198963080] mb P  I16..4: 11.0% 35.2%  8.1%  P16..4: 21.9% 14.0%  7.5%  0.0%  0.0%    skip: 2.3%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] mb P  I16..4: 11.0% 35.2%  8.1%  P16..4: 21.9% 14.0%  7.5%  0.0%  0.0%    skip: 2.3%
[libx264 @ 0x7d7198963080] mb B  I16..4:  2.5%  3.6%  0.7%  B16..8: 38.9% 19.5%  4.3%  direct:16.7%  skip:13.8%  L0:43.2% L1:40.0% BI:16.8%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] mb B  I16..4:  2.5%  3.6%  0.7%  B16..8: 38.9% 19.5%  4.3%  direct:16.7%  skip:13.8%  L0:43.2% L1:40.0% BI:16.8%
[libx264 @ 0x7d7198963080] final ratefactor: 11.73
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] final ratefactor: 11.73
[libx264 @ 0x7d7198963080] 8x8 transform intra:62.1% inter:59.8%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] 8x8 transform intra:62.1% inter:59.8%
[libx264 @ 0x7d7198963080] coded y,uvDC,uvAC intra: 52.5% 40.1% 37.7% inter: 43.1% 23.6% 8.6%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] coded y,uvDC,uvAC intra: 52.5% 40.1% 37.7% inter: 43.1% 23.6% 8.6%
[libx264 @ 0x7d7198963080] i16 v,h,dc,p: 52% 29% 11%  8%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] i16 v,h,dc,p: 52% 29% 11%  8%
[libx264 @ 0x7d7198963080] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 22% 25% 31%  4%  4%  2%  4%  2%  6%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 22% 25% 31%  4%  4%  2%  4%  2%  6%
[libx264 @ 0x7d7198963080] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 27% 27% 20%  5%  6%  3%  4%  3%  4%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 27% 27% 20%  5%  6%  3%  4%  3%  4%
[libx264 @ 0x7d7198963080] i8c dc,h,v,p: 77% 12%  9%  2%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] i8c dc,h,v,p: 77% 12%  9%  2%
[libx264 @ 0x7d7198963080] Weighted P-Frames: Y:11.5% UV:3.2%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] Weighted P-Frames: Y:11.5% UV:3.2%
[libx264 @ 0x7d7198963080] ref P L0: 62.3%  8.5% 19.6%  9.2%  0.4%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] ref P L0: 62.3%  8.5% 19.6%  9.2%  0.4%
[libx264 @ 0x7d7198963080] ref B L0: 87.9%  9.6%  2.5%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] ref B L0: 87.9%  9.6%  2.5%
[libx264 @ 0x7d7198963080] ref B L1: 96.0%  4.0%
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] ref B L1: 96.0%  4.0%
[libx264 @ 0x7d7198963080] kb/s:19809.02
15:55:07 [DEBUG] (2) gyroflow::rendering: [libx264 @ 0x7d7198963080] kb/s:19809.02
15:55:07 [DEBUG] (2) gyroflow_core::filesystem: FfmpegPathWrapper::drop file:///mnt/aaa/bbb/medias/20240407_140957956.TS.mp4
```

AdrianEddy, 2024-04-27:
Try to disable audio export
Which camera is the video from? The error message indicates that it has some corrupted frames

Surfoo, 2024-04-27:
Same result without the audio.
The video come from a smartphone, Pixel 6a.

AdrianEddy, 2024-04-27:
can you send me the original video file?

Surfoo, 2024-04-27:
Yes, to which email? with a wetransfer link?

AdrianEddy, 2024-04-28:
just send the wetransfer link here, there's an option to get a link without email there

Surfoo, 2024-04-28:
Here the link: https://we.tl/t-DgDLKRLQIr It's an other video but with the same error.

AdrianEddy, 2024-04-28:
Fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2024-04-28 by AdrianEddy

Surfoo, 2024-04-28:
Wow, you are awesome! 

Can you explain me the bug?

AdrianEddy, 2024-04-28:
that particular file has duplicated packet timestamps, and ffmpeg sometimes tolerates that, but sometimes not.
This is specific to your recording device and how it encodes things, the bug is actually there. I just made Gyroflow ignore that issue instead of aborting entire render

Changed lines: https://github.com/gyroflow/gyroflow/commit/67105d1133ac467817cef88e042514860c9da276