gyroflow/gyroflow #827: 2d Optical-flow stabalization

siefkenj, 2024-04-30:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I have some video without a lens profile and without gyro data. I find that OpenShot's 2d-only stabilization produces smoother and less-distorted results than Gyroflow's. However OpenShot's hardware acceleration doesn't work. It would be really great if Gyroflow had a setting to recreate this simple stabilization mode.

The algorithm seems fairly simple, and is here: https://github.com/OpenShot/libopenshot/blob/develop/src/CVStabilization.cpp
It looks to use FlowPyrLK data and then OpenCV's `estimateAffinePartial2D` to find displacement/rotation vectors.

Would this be easy to implement? If someone can point me towards the right place in the sources, I could try to implement it.

closed 2024-04-30 by AdrianEddy

siefkenj, 2024-04-30:
@AdrianEddy Perhaps you missed the _2d_ label on this issue? Currently when doing optical-flow stabilization Gyroflow seems to rotate in 3 dimensions to try and stabilize the video. I am hoping for stabilization done by only `x`/`y` translation and `roll` adjustments.

AdrianEddy, 2024-04-30:
the 2d part is the reason why #45 is still open. Quote from my comment there: `Proper implementation should focus more on better optical flow tracking, and there should be a second optical flow stabilization on top of it (using motion vectors to move small pixel groups, like Warp Stabilizer, instead relying only on estimated camera motion).`