gyroflow/gyroflow #828: Export-Folderaccess under MacOS

morbificagent, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

v1.5.4

### What operating system are you using?

macOS Monterey 12.6.7

### What happened?

When i try to start a render queue i get an errormessage:

**ffmpeg error: ffmpeg::Error(1: Operation not permitted)**

It looks like its because of a missing "Manual" selection of the export-folder because if i dont use the render queue and try to export a single video gyroflow notifies me to "Manual select the exportfolder because of macos restrictions".... After i do this in the following popup its working for a single file.
But if i use the render queue Gyroflow doesnt ask me about this and fails with the shown errormessage.

Gyroflow is installed via Mac Appstore. I have tried to give gyroflow full disk access but this didnt helped.

Many thanks

morbificagent, 2024-05-02:
OK, tested the "Testflight-Version":
When i try to add files to the queue nothing happens after selecting the files and click "add"...

Tested the dmg file from https://nightly.link/gyroflow/gyroflow/workflows/release/master/Gyroflow-macos.zip:
Its working fine!

AdrianEddy, 2024-05-05:
> When i try to add files to the queue nothing happens after selecting the files and click "add"...

Can you clarify that? Where are you selecting files, and which "Add"?

AdrianEddy, 2024-05-24:
This is fixed in latest TestFlight build

closed 2024-05-24 by AdrianEddy