gyroflow/gyroflow #829: BRAW crashing when loading new file after successfully loaded another one

pkarc, 2024-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4 (gh1869)

### What operating system are you using?

Windows 11

### What GPU are you using?

NVIDIA GeForce GTX 1650 Ti

### What happened?

I'm exporting project process data directly from the UI using BRAW file, the program crash when loading a BRAW file after successfully loading and exporting proccess data of the 1st successfully loaded file, it happens randomly sometimes the 2nd, 3rd, 4th, etc...

### Relevant log output

```shell
23:10:46 [INFO] [MDK] 0.26.0 (git 132d28a) - Multimedia Development Kit. Copyright (c) 2016-2024 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600/10.0.22631 x86_64; Clang18.1.3 (++20240322073153+ef6d1ec07c69-1~exp1~20240322193300.86)MSVC1933; vcruntime14.28.29333.0/msvcp140.dll; 15:19:45 Apr 30 2024
23:10:46 [DEBUG] (1) gyroflow: [in main] Qt version: 6.7.0
23:10:46 [DEBUG] (1) gyroflow: Gyroflow 1.5.4 (gh1869)
23:10:46 [INFO] Videos: ["D:\\A001_04290218_C016.braw"]
23:10:46 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\ivand\\Downloads\\Gyroflow-windows\\Gyroflow-windows64\\camera_presets"
23:10:47 [INFO] Loaded 14724 lens profiles in 443.089ms
23:10:47 [INFO] OpenCL Platform: NVIDIA CUDA, ext: Extensions { inner: "cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_d3d10_sharing cl_khr_d3d10_sharing cl_nv_d3d11_sharing cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_win32 cl_khr_external_memory_win32" } Device: NVIDIA Corporation NVIDIA GeForce GTX 1650 Ti
23:10:47 [INFO] Custom decoder: BRAW:gpu=auto
23:10:47 [INFO] [MDK] default000002420E775680 virtual void __cdecl mdk::FrameReader::stop(void), current state: 0
23:10:47 [INFO] [MDK] default000002420E775680 virtual void __cdecl mdk::FrameReader::stop(void), current state: 0
23:10:47 [INFO] [MDK] 000002420EA53D80 MediaControl.prepare(0, ...) D:/A001_04290218_C016.braw
23:10:47 [INFO] [MDK] 000002420E807200 url opened
23:10:47 [INFO] [MDK] Format: mov,mp4,m4a,3gp,3g2,mj2, range: 0 +48883ms, bitrate: 812334654, size: 0
Metadata:
  braw_compression_ratio: 8:1
  creation_time: 2024-04-29T02:18:12.000000Z
  camera_type: Blackmagic Pocket Cinema Camera 4K
  post_3dlut_embedded_name: Blackmagic Gen 5 Film to Extended Video.cube
  firmware_version: 8.1
  shot_type: None
  post_3dlut_embedded_title: Gen 5 Film to Extended Video
  encoder_device_manufacturer: Blackmagic Design
  environment: interior
  sensor_line_time: 7.507507
  post_3dlut_mode: Disabled
  post_3dlut_embedded_bmd_gamma: 
  post_3dlut_embedded_data: 0
  viewing_gamma: Blackmagic Design Film
  viewing_gamut: Blackmagic Design
  manufacturer: Blackmagic Design
  camera_id: 5aa1da89-5ff8-47f0-8856-5e774c76db39
  shutter_type: Angle
  time_lapse_interval: 
  camera_operator: 
[... 525 lines cut ...]
23:11:04 [DEBUG] (1) mdk: [MDK] default00000157545FA7A0 deleting FrameReader...
23:11:04 [DEBUG] (1) mdk: [MDK] player is destroyed
23:11:04 [DEBUG] (1) mdk: [MDK] __cdecl mdk::VideoRenderer::Private::~Private(void) @127
23:11:04 [DEBUG] (1) mdk: [MDK] ~~~~~~~releasing cuda resources. stream: 000001572C969630, pinned host memory: 0000000000000000
23:11:17 [DEBUG] (122) mdk: [MDK] default 00000157545F8DC0 FrameReader::update MediaStatus 0X114=>0X154
23:11:17 [DEBUG] (122) mdk: [MDK] read packet error. FrameReader@00000157545F8DC0 sending Packet::End to all packet queues. PacketIO@0000015723EA1DB0 ->url: D:/A001_04290218_C016.braw
23:11:19 [DEBUG] (2) gyroflow_core::camera_identifier: Camera ID Brand: Blackmagic, Model: Pocket Cinema Camera 4K, Metadata: Object {"camera_type": String("Blackmagic Pocket Cinema Camera 4K"), "firmware_version": String("8.1"), "viewing_bmdgen": Number(5), "crop_origin": Array [Number(16.0), Number(8.0)], "crop_size": Array [Number(4096.0), Number(2160.0)], "safe_area": Array [Number(0.0), Number(0.0)], "sensor_area_captured": Array [Number(4096.0), Number(2160.0)], "braw_compression_ratio": String("8:1"), "braw_codec_bitrate": Number(813189120), "encoder_device_manufacturer": String("Blackmagic Design"), "post_3dlut_mode": String("Disabled"), "post_3dlut_embedded_name": String("Blackmagic Gen 5 Film to Extended Video.cube"), "post_3dlut_embedded_title": String("Gen 5 Film to Extended Video"), "post_3dlut_embedded_bmd_gamma": String(""), "post_3dlut_embedded_size": Number(33), "viewing_gamma": String("Blackmagic Design Film"), "viewing_gamut": String("Blackmagic Design"), "sensor_line_time": Number(7.50750732421875), "ois_enable": Number(0), "manufacturer": String("Blackmagic Design"), "camera_id": String("5aa1da89-5ff8-47f0-8856-5e774c76db39"), "offspeed": Number(0), "shutter_type": String("Angle"), "time_lapse_interval": String(""), "frameguide_aspect_ratio": String(""), "anamorphic_enable": Number(0), "anamorphic": String(""), "clip_number": String("A001_04290218_C016"), "reel_name": String("1"), "scene": String("1"), "shot_type": String("None"), "take": String("59"), "take_type": String("None"), "good_take": String("false"), "environment": String("interior"), "day_night": String("day"), "lens_type": String(""), "filters": String(""), "production_name": String(""), "director": String(""), "camera_number": String("A"), "camera_operator": String(""), "date_recorded": String("2024:04:29"), "location": String(""), "rotation": Number(0), "analog_gain": Number(1.0), "analog_gain_is_constant": Number(1)}
23:11:19 [DEBUG] (2) gyroflow_core::camera_identifier: Camera ID Brand: Blackmagic, Model: Pocket Cinema Camera 4K, Metadata: Object {"sensor_rate": Array [Number(60), Number(1)], "shutter_value": String("180°"), "internal_nd": Number(0.0), "analog_gain": Number(1.0), "as_shot_kelvin": Number(0), "as_shot_tint": Number(0), "aperture": String(""), "exposure": Number(0.0), "focal_length": String(""), "distance": String(""), "iso": Number(1000), "white_balance_kelvin": Number(4000), "white_balance_tint": Number(65526)}
23:11:19 [INFO] CameraIdentifier {
    brand: "Blackmagic",
    model: "Pocket Cinema Camera 4K",
    lens_model: "",
    lens_info: "",
    focal_length: None,
    camera_setting: "",
    fps: 60000,
    video_width: 4096,
    video_height: 2160,
    additional: "",
    identifier: "",
}
23:11:19 [DEBUG] (2) telemetry_parser::util: Reading duration: (None, None, None)
23:11:19 [DEBUG] (2) gyroflow_core::filesystem: FileWrapper::drop file:///D:/A001_04290218_C016.braw
23:11:19 [WARN] IMU duration 48883.88298048531 is different than video duration (48883)
```

AdrianEddy, 2024-05-04:
Should be fixed in latest dev build https://gyroflow.xyz/devbuild/

closed 2024-05-10 by AdrianEddy

alexrjohnston, 2025-05-05:
This is now happening to me for some reason - 1.6.1 M1 mac pro

first file - loads fine, second file gyroflow freezes. Looks like the fix was added in 1.6.0, what am I missing?