gyroflow/gyroflow #830: Trimm settings is not working correctly from created settings preset

AndriiShramko, 2024-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4 gh1873

### What operating system are you using?

Windows 10 Pro 22h2

### What GPU are you using?

gtx1080 ti

### What happened?

Trimm settings is not working correctly from created settings preset. It is a different cameras from a 360 camera rigand I need trimm each file exactly the same.
As you can see on a screenshots - In point is always different, but need to be the same as I am using a Trimrange in preset.
Please check it. Let me know and I can pay you for you work. I need it as much as possible. Thanks Best Regards Andrii Shramko

https://drive.google.com/file/d/1mH1FMzPP-c6OU__-uyc50leZ5_rohxGE/view?usp=sharing preset
[image]
[image]
[image]

AdrianEddy, 2024-05-09:
Is the duration of all these files the same? If not, are they starting at the same time, but ending at different times?

AndriiShramko, 2024-05-09:
Hello
no. It has a different duration. Different start and different end. As I
understanding - if I use trimm inpoint at 1000frames is must trimm all
video at  in point  of 1000frames, but it alway differen.

AdrianEddy, 2024-05-09:
The way it works is it stores the trim range as a percentage of the input, so it's affected by the duration as well.
If they also don't start at the same time, then you need to synchronize them in 360 software anyway. so can't you just synchronize after trimming?

AndriiShramko, 2024-05-09:
The ideal is to synchronize all cameras by gyro data and render the right
chunks of all synchronized videos from the long video. People was told that
this is very difficult to implement, so I use the old-fashioned way, using
a loud clap at the beginning of the video, which would then synchronize by
sound. I always rendered the selected piece of video with the original
video length, but with black frames. This was very time consuming, so I
decided to use the trimm functionality. But since it uses a percentage of
the video length and not a specific number of frames, the cut pieces of
video have an unpredictable start. And it is impossible to synchronize in
other programs, if you don't render the whole video, because only at the
beginning of the video there is a loud sound for video synchronization. I
create professional 360 cameras for big companies and I am looking for
solutions to make the post-production phase as easy as possible. Gyroflow
is a magic program. But now it is difficult to adapt it for video
processing from multi-camera 360 rig. I'm now coming up with a workflow and
using audio sync in davinchi and stabilization through Gyroflow plugin. I
think it will work, but the option of synchronization of cameras by gyro
data I will also plan to implement, if I find a specialist who wants to do
this. :)
And I think it is better to use a frame numbers, than a percentage for
trimming tool. Any way you've made a great tool! Thank you for your willing
to help with that.

AdrianEddy, 2024-05-09:
you could write a script that will go through the .gyroflow project files and calculates the correct trim range based on the duration.

Changing the trim range to use frames instead of the percentages is much more work than you think, it needs to take into account current project files, implement compatibility between old and new approach, and the trim range values use percentages across tens of different areas and modules in gyroflow, so all of these would have to be checked too after this change.

and I'm not sure that storing the trim range as a frame number is correct either. what if you have different frame rates? There are other use cases than 360 videos, and I'm pretty sure I'll get another issue about it using frame numbers from someone else for which this doesn't work.  It's not easy to cover all people's use cases.

Storing trim rage in seconds could be the best idea, but still the first point remains, and it's not easy to do without breaking something else

AdrianEddy, 2024-05-09:
Using the plugin would probably be the best approach, you simply load the same project file in plugin for all your videos

AdrianEddy, 2024-05-10:
ok I changed it to store as milliseconds, try the dev build https://gyroflow.xyz/devbuild/

closed 2024-05-10 by AdrianEddy

AndriiShramko, 2024-05-14:
Great! Thank you!
Now it works OK.

But I have another problem with cropping Aeria
https://www.youtube.com/watch?v=0tvtK5BSvWM