gyroflow/gyroflow #833: Motion blur magic

AdrianEddy, 2024-05-10:
We could implement a way to remove the motion blur using AI model: https://github.com/google-research/maxim

Then using the new AI-based optical flow, we can add the motion blur back after stabilization

For reference:
- https://github.com/bevyengine/bevy/pull/9924/files
- https://github.com/phiresky/opencl-motion-interpolation
- https://github.com/OpenCL/GEGL-OpenCL/blob/master/opencl/motion-blur-linear.cl

webben216, 2024-05-25:
If I understand correctly, do you want to process all frames with the neural network or only certain frames with a certain amount of blur? The example on replicate takes around 17 seconds for a single frame on a Nvidia T4 GPU, im my opinion way to long to process all frames.

Looking at my footage with long exposure times (walking), only around 10% of the frames have significant motion blur.

Things I am asking myself:
- How bad is the flickering on successive frames? Do we need some temporal filtering?
- How does the model deal with 10, 12, or 14 bit raw/ log footage?

AdrianEddy, 2024-05-25:
Only some frames, the ones with most blur, based on the movement amount from gyro data. Each frame will likely need to be stored to a file. Hard to tell exactly how it will behave in real world, but I imagine we can add some blending or something if there's too much inconsistency.

I'm sure the network will be much faster in Rust than in Python. Python does **a lot** of slow things before and after actual neural network, we'll do all of them on a GPU in a single step instead

webben216, 2024-05-25:
That makes sense. I never used cog, but with TensorRT, the inference time didn't change significantly between the Python and C++.

Why do we need to add motion blur back in Gyroflow? In my view, motion blur is a totally separate feature. The only advantage I see adding motion blur in Gyroflow is a reduced crop, if it is possible to apply it to the full unstabilized frame with the information of the stabilization.

AdrianEddy, 2024-05-25:
Because motion blur looks good if it's done right. That's why people use ND filters and slow shutter speeds, but this is incompatible with digital stabilization, so we can add it artificially after stabilization.

webben216, 2024-06-15:
I made series of test shots with different exposure angles in different situations and tested frames using the Replicate inference. The results look really well, especially in region with high contrast and a lot of details.

I see two main use cases for this feature: 
1. Removal of motion blur in low light shots. For low light shots people are forced to shot with long exposure times and a general reduction of motion blur is intended.

2. People, who want to shoot in daylight and want motion blur in their shots (biking, fpv) with the best quality possible. For them, only the rotational motion blur is bad, not the translational motion blur (which can be archived with nd filters).

My idea for use case 2:
- use optical flow to create a map for the motion blur, ideally distinguishing between rotational and translational movement
- apply the deblurring model to the frame
- use the map to blend between the deblurred and the original frame

This way we could keep the high quality translational motion blur at the edges while reducing the rotational motion blur in the center, which causes these disruptive bumps.

**This is a sample image with a 180 degree shutter angle:**
[image]

**This is the deblurred image:**
[image]

VladimirP1, 2024-10-15:
I think it should be possible to remove **rotational** blur (case 2) purely based on gyro data without any AI. We can certainly estimate the point spread function caused by rotational motion at any point in the image. I am still not sure how to solve for the estimated image after finding the PSFs, but I know it should be possible.

noobtoob4lyfe, 2024-11-21:
The problem is that the blur removal is not good enough to recover the the thin branches that basically become invisible because of the blur.  Those thin features disappearing and reappearing are the most egregious signs that you stabilized footage with too much motion blur.  So the juice is not really worth the squeeze here

AdrianEddy, 2024-11-21:
I think it is definitely worth the squeeze. Sure it won't be perfect, but it will be better

AdrianEddy, 2025-01-09:
> I think it should be possible to remove **rotational** blur (case 2) purely based on gyro data without any AI. We can certainly estimate the point spread function caused by rotational motion at any [...]

@VladimirP1 I thought so too, but I can't get it working for some reason. I created a branch `motion-blur` which implements sampling different points according to IMU data, but it doesn't work as expected. I think something is wrong with choosing the quaternion per shutter sample

https://github.com/gyroflow/gyroflow/blob/motion-blur/src/core/stabilization/frame_transform.rs#L203-L234
https://github.com/gyroflow/gyroflow/blob/motion-blur/src/core/gpu/opencl_undistort.cl#L596

(only OpenCL kernel was updated)

for reference, all changes vs master: https://github.com/gyroflow/gyroflow/compare/master..motion-blur

VladimirP1, 2025-01-11:
@AdrianEddy but it indeed seems to add motion blur, that's what I would expect it to do (but chroma seems to be broken, maybe that's because of intel opencl on linux)

[image]

VladimirP1, 2025-01-11:
Removing motion blur seems to be much harder in our case, because PSF varies in different parts of a frame. So the problem to be solved here is called "spatially varying deconvolution" and it is much harder than simple deconvolution with one PSF.

FloppySoft, 2025-01-28:
> We could implement a way to remove the motion blur using AI model: https://github.com/google-research/maxim
> 
> Then using the new AI-based optical flow, we can add the motion blur back after stabilization
> ...

Is AI really necessary? I always assumed the gyro data along with lens correction data will suffice to calculate / estimate motion vectors of each pixel during that shot frame (to counter motion blur by camera movement, not other motion blur).

I found old papers at awesome-deblur
https://github.com/subeeshvasu/Awesome-Deblurring?tab=readme-ov-file#multi-imagevideo-motion-deblurring
like this one
https://cv.snu.ac.kr/publication/jour/2018/thkim_pami2018_dynamic.pdf
which appears to me to be able to create results being "worth the queeze", doesn't it?

 Photoshop is able to directionally sharpen via "smart sharpen" and I always assumed that would be a conventional algorithm.

Regarding whether it is worth it: I find stabilized motion blur quite disrupting. It happens suddenly and appears like a strong interference. Hence, I even bought an action cam that is able to somehow handle low light and motion blur a little better. I removed motion blur from many photos myself and found the results to contain more perceivable details than the blurry original. I can only assume that motion denoise would at least reduce the visual impact, assumed that the footage is no strongly shaking action footage, so use case would rather be walking through dark environments.

If people want to add pixel based motion blur, they could use conventional video editors to do that afterwards.