gyroflow/gyroflow #834: Broken on Nobara Linux

OttCS, 2024-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

Linux (Nobara 39 Gnome Edition)

### What GPU are you using?

NVIDIA 1660 ti

### What happened?

Installation fails on my system (Nobara, which is a Fedora derivative), whether I'm using the .AppImage or the .tar.gz.

When attempting to run with the .AppImage, I allow the file to be "Executable as Program" through the file properties panel, and then attempt to run. It simply produces a .dmp file (named "e32f007c-e098-4fdf-180228ae-72b4b9d9" or something similarly cryptic) and nothing else occurs.

When attempting to run with the .tar.gz, I extract and go in the Gyroflow directory (with the gyroflow executable). When I run ./gyroflow here, I recieve the following feedback:

```
./gyroflow: error while loading shared libraries: libc++.so.1: cannot open shared object file: No such file or directory
```

I did attempt to install the **possibly needed packages** as well as the GPU specific packages (after changing `apt` to `dnf`), but most aren't found:

```
Last metadata expiration check: 0:55:46 ago on Sun 12 May 2024 03:36:19 PM EDT.
No match for argument: libva2
No match for argument: libvdpau1
No match for argument: libasound2
No match for argument: libxkbcommon0
No match for argument: libpulse0
No match for argument: libc++-dev
No match for argument: libvulkan1
Error: Unable to find a match: libva2 libvdpau1 libasound2 libxkbcommon0 libpulse0 libc++-dev libvulkan1
```

OttCS, 2024-05-12:
Distributing the app through Flathub may work for easier Linux installations and updates, and could clear up issues with dependencies: https://docs.flathub.org/docs/for-app-authors/why-flathub

AdrianEddy, 2024-05-12:
https://superuser.com/a/1760413

I don't know anything about flatpak but I'll be happy to accept PRs

closed 2024-05-12 by AdrianEddy