gyroflow/gyroflow #835: smoothed gyro data or final applied rotation matrics for image stabilization

Ming-chwan, 2024-05-15:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I just want use the smoothed gyro data or saying , the final rotation matrics for video rotation  before GPU rending, any method to dump these information ? Thanks

AdrianEddy, 2024-05-15:
Yes, just use the .gyroflow project file (save as "Project file (with processed gyro data)".
See also https://github.com/EmberLightVFX/Gyroflow-to-CSV

closed 2024-05-15 by AdrianEddy

Ming-chwan, 2024-05-17:
Thanks AdrianEddy,  another doubt is,  the script will dump (timestamp, quaternions) info into a file , does it represent the content  of varible  **smoothed_quat1** in code fragment :  let smoothed_quat1:Unit<Quaternion<f64>> = gyro.smoothed_quat_at_timestamp(timestamp_ms); ?.  
 
if it is YES,  I  just use the smoothed quaternions (rotation ) to generate a rotation matric in my program to rotate my video frame as it does in frame_transofrm.rs.    
Thanks!

AdrianEddy, 2024-05-17:
[Yes](https://github.com/gyroflow/gyroflow/blob/master/src/core/lib.rs#L1041-L1058)