gyroflow/gyroflow #838: Wrong "dynamic zooming" russian translation

lukegluke, 2024-05-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

Any

### What happened?

Dear @AdrianEddy, quite time ago on crowdin we fixed wrong copypaste russian translation for "Dynamic zoom" mode and  you approved the correct one (Динамический зум = Dynamic zoom): https://crowdin.com/editor/gyroflow/7/en-ru?view=comfortable#2633

But in 1.5.4 application there is still this issue with wrong translation:
[image]
(=No zoom, No zoom, Static zoom instead of No zoom, Dynamic zoom, Static zoom)

AdrianEddy, 2024-05-18:
Yes because there was no official update since 1.5.4. use the dev build https://gyroflow.xyz/devbuild

closed 2024-05-18 by AdrianEddy

lukegluke, 2025-03-18:
Dear @AdrianEddy,

Found another small issue with Russian translation for "Output size" in Export. The translation itself is right, but it is too long so input fields of resolution don't show whole number (on default width of side panel):
[image]

I've added a short version "Разрешение" = "Resolution", could you please apply it
https://crowdin.com/editor/gyroflow/7/en-ru?view=comfortable#1313

And also there could  be a reason to set/increase minimal width for these and maybe other input fields so no translation on any language will break them again.

AdrianEddy, 2025-03-19:
updated in latest dev build