gyroflow/gyroflow #845: Core i5-4570t not runing gyroflow

Grindio1966, 2024-06-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Do you have latest GPU drivers installed?

- [X] I have the latest GPU drivers installed, I'm absolutely sure.

### Have you checked the documentation?

- [X] I have read the FAQ and Troubleshooting and didn't find my issue there.

### Gyroflow version

1.5.4

### What operating system are you using?

Windows 10

### What GPU are you using?

Intel® HD Graphics 4600

### What happened?

I can't export files in gyroflow, it will open the file, but when I press "export" gyroflow blocks, the cpu is a core i5-4570t, not sure if it is supported. I've already tried h264 and disabling gpu encoding but to no avail.

Grindio1966, 2024-06-14:
Well I just found a cure for this, as I read in a previous post I edited the following file:
Edit "Gyroflow_with_console.bat" and add this as a second line:
set NO_OPENCL=1
also add : 
set NO_WGPU=1 as a third line
then run Gyroflow_with_console.bat
It's working but slooow just 0.6 fps, my samsung s6 lite is way faster

AdrianEddy, 2024-06-14:
You need to install the drivers for your GPU, download from intel.com

closed 2024-06-14 by AdrianEddy