gyroflow/gyroflow #848: is support instra360 x3 action camera?

cuixing158, 2024-06-26:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I read the official documentation to understand that insta360 (OneR, OneRS, SMO 4k, Go, GO2, GO3, Caddx Peanut, Ace, Ace Pro) is supported, meaning that x3 products are not supported? But I imported the insv video recorded by x3 into gyroflow app, and I can see that a lot of information is automatically inferred, including camera configuration parameters, internal parameters, distortion coefficients, distortion centers, gyro curves, and the question is that if it's supported, why is it that I'm not de-distorting the effect of verifying the image with the camera's configuration parameters using an opencv script? My sample code is below:

[image]

```python
import numpy as np
import cv2

# 从gyroflow软件推测得到
K = np.array([[2304.81, 0, 1219.724], [0, 2304.905, 1444.969], [0, 0, 1]])
D = np.array([0.40000504, 1.19084167, -3.77110991, 0.00022223])

vi = cv2.VideoCapture(
    "/mnt/f/allData/insta360_videos/20240612/VID_20240612_101125_00_009.insv"
)
vi.set(cv2.CAP_PROP_POS_FRAMES, 100)
res, frame = vi.read()
print("frame.shape:{}".format(frame.shape))

# undistortImg = cv2.fisheye.undistortImage(frame, K, D)
mapX, mapY = cv2.fisheye.initUndistortRectifyMap(
    K, D, np.eye(3), K, frame.shape[:2], cv2.CV_32FC1
)
undistortImg = cv2.remap(frame, mapX, mapY, cv2.INTER_LINEAR)

undistortImg = cv2.resize(undistortImg, (512, 512))
cv2.imshow("", undistortImg)
cv2.waitKey()
```

The output of the code is still a circular fisheye distorted image, what is the problem?

AdrianEddy, 2024-06-26:
Gyroflow doesn't support 360° cameras and the support is not planned at the moment.
The reasons are:
1. It requires stitching algorithm which is not easy to do univerally
2. It requires different lens distortions model
3. It requires to read 2 video streams at once, which is currently not possible with the libraries we use

The effort to make 360° support in Gyroflow is too big and I'm not planning to work on it. However, if someone provides a PR, I'll probably accept it, but it's a lot of work

closed 2024-06-26 by AdrianEddy

Erudition, 2024-10-03:
We don't need to read 2 streams or do stitching manually if we work on the video output by insta360 studio rather than raw *.insv files. The Insta360 flow stabilization can be turned off, too.

In other words, export a single video of otherwise raw, unstabilized 360 footage, then the only thing to support is the 360 lens distortion model as @AdrianEddy says.

AdrianEddy, 2024-10-03:
yeah but how useful is using gyroflow if you already have to use Insta360 studio? That seems pointless to me, their own stabilization is good too

Erudition, 2024-10-15:
You're right, the Flow Stabilization in Insta360 Studio is good. The main difference is that it has no controls whatsoever, and the motion data can't be used for other purposes.

hilmiyafia, 2024-10-16:
@AdrianEddy I agree with @Erudition Insta360 stabilization is good, but the choice is either fully stabilized or not. I think it would be amazing if there's a slider to control how much stabilization we want like the one that Gyroflow currently has.

So instead of fully stabilized, it will still follow the forward direction, but smoother.

We don't need to do the stitching in Gyroflow. Gyroflow can just read equirectangular videos that already stitched from Insta360 app/studio. Then load the motion data from the original insv file.